verilator/verilator #4144: Cannot declare static variable with same name in different blocks of method

tudortimi, 2023-04-23:
I have the following code:

```
// file: test.sv

module test;

  function do_stuff();
    begin: block0
      static int some_int;
    end
    begin: block1
      static int some_int;
    end
  endfunction 

endmodule
```

When I compile with `verilator --binary test.sv` I get:

```
%Error: test.sv:8:18: Duplicate declaration of signal: 'test.do_stuff__Vstatic__some_int'
    8 |       static int some_int;
      |                  ^~~~~~~~
        test.sv:5:18: ... Location of original declaration
    5 |       static int some_int;
      |                  ^~~~~~~~
%Error: Exiting due to 1 error(s)
```

It seems that `some_int` is put under `do_stuff` directly, not under `do_stuff.block0` and `do_stuff.block`.

tudortimi, 2023-04-23:
I'm using v5.008-143-gcac634d39

wsnyder, 2023-04-24:
V3Begin should be renaming to avoid these sort of issues.  Might you be willing to make this into a test_regress test case and debug?

tudortimi, 2023-04-24:
@PeterMonsson is this something you'd be interested in doing?

PeterMonsson, 2023-04-26:
@tudortimi yes, but I am not sure that I get to it the next two weeks.

tudortimi, 2023-04-26:
It's not blocking for us w.r.t. SVUnit, but I found it in that context. Also, I think @wsnyder doesn't want to leave it hanging.

closed 2023-05-03 by wsnyder

PeterMonsson, 2023-05-03:
Thank you @wallento

commit 5ea201d0 (2023-05-02): Duplicate static names in blocks in functions

Static variables of functions are created in the function. When blocks
in a function use identical names for static variables, we need to name
those variables properly.

Fixes #4144
commit 670b7d61 (2023-05-02): Duplicate static names in blocks in functions

Static variables of functions are created in the function. When blocks
in a function use identical names for static variables, we need to name
those variables properly.

Fixes #4144
commit 295ef0a4 (2023-05-02): Fix duplicate static names in blocks in functions

Static variables of functions are created in the function. When blocks
in a function use identical names for static variables, we need to name
those variables properly.

Fixes #4144
PR #4160 (2023-05-03): Fix duplicate static names in blocks in functions (#4144)
Static variables of functions are created in the function. When blocks in a function use identical names for static variables, we need to name those variables properly.

Fixes #4144