verilator/verilator #4147: Strange crashes with Verilator in code that otherwise compiles and runs

tudortimi, 2023-04-24:
I have some internal crashes during compilation for code that otherwise runs and compiles. I run multiple tests against the source code, but about 5 of the tests end in an internal error. I can't figure out a pattern either.

I get something like the following:

```
- V3Ast.cpp:1215:     Dumping obj_dir/Vtestrunner_012_width.tree
- V3Width.cpp:7477:   widthCommit: 
- V3Const.cpp:3751:   constifyAllLive: 
- V3Ast.cpp:1215:     Dumping obj_dir/Vtestrunner_014_const.tree
%Error: Internal Error: /home/tudor/projects/github/svunit/svunit/svunit_base/svunit_testcase.sv:95:9: ../V3Broken.cpp:186: No dtype on node with hasDType(): NEW 'new'
                                                                                                     : ... In instance testrunner.___05Fts
-node: NEW 0x213d9b0 <e15116> {n95aj} @dt=0@  new pkg=0x2135b00 -> FUNC 0x20e1be0 <e20787> {m56ax} @dt=0x207cd30@(w0)void  new [METHOD]
   95 |   super.new(name);
      |         ^~~
                        /home/tudor/projects/github/svunit/svunit/svunit_base/svunit_base.sv:86:1: ... note: In file included from svunit_base.sv
                        /home/tudor/projects/github/svunit/svunit/svunit_base/svunit_string_utils.svh:25:1: ... note: In file included from svunit_string_utils.svh
                        /home/tudor/projects/github/svunit/svunit/svunit_base/svunit_types.svh:31:47: ... note: In file included from svunit_types.svh
                        /home/tudor/projects/github/svunit/svunit/svunit_base/svunit_pkg.sv:24:45: ... note: In file included from svunit_pkg.sv
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1215:     Dumping obj_dir/Vtestrunner_990_final.tree
- V3Broken.cpp:330:   Broken called under broken, skipping recursion.
- V3StatsReport.cpp:226:statsReport: 
%Error: Internal Error: Aborting since under --debug
%Error: export VERILATOR_ROOT=
%Error: ulimit -s unlimited 2>/dev/null; exec /opt/verilator/v5.008-143-gcac634d39/bin/verilator_bin_dbg --debug --binary --top-module testrunner -f .svunit.f +define+SVUNIT_VERSION=\"SVUnit\ foo\"
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /opt/verilator/v5.008-143-gcac634d39/bin/verilator_bin_dbg --debug --binary --top-module testrunner -f .svunit.f +define+SVUNIT_VERSION=\"SVUnit\ foo\"
```

The code mentioned in the error message is legal SystemVerilog code. It compiles in all big 3 simulators and it also compiles in Verilator. It's just some regular constructor calling a `super.new()`.

Any pointers on how to debug this?

wsnyder, 2023-04-29:
It sounds like some variable isn't properly initialized so random state (or random data layout) makes a problem.

Try configuring and compiling verilator with:

    ./configure CXX="g++ -fsanitize=address,undefined"
    
 and run with --debug (as you did) and also --gdbbt and see if it gives more information.

tudortimi, 2023-05-02:
On it.

tudortimi, 2023-05-02:
I'm trying to understand what this code is supposed to do. From the comment, I'm guessing it's supposed to look for code like `$display($sformatf(...), $sformatf(...))` and merge it into `$display($sformatf(...))`?

Even though I don't have any of that kind of code, I assume that some other construct could cause this code to "activate" in an unexpected way. I have some code like this, which seems similar:

```
// A macro
`define INFO(msg) \
  $display("INFO:  [%0t][%0s]: %s", $time, name, msg);

// Usage of that macro
`INFO($sformatf(`"%s::PASSED`", _testName));
```

This would expand to something like:

```
$display("INFO:  [%0t][%0s]: %s", $time, name, $sformatf(`"%s::PASSED`", _testName));
```

Could this trip up the code listed above?

tudortimi, 2023-05-02:
I tried it out in a small example a try and there it worked:

```
module test;
  
  initial begin
    do_stuff();
    $finish();
  end

  function automatic void do_stuff();
    string name = "some_name";
    string _testName = "some_test";
    $display("INFO:  [%0t][%0s]: %s", $time, name, $sformatf(`"%s::PASSED`", _testName));
  endfunction

endmodule
```

wallento, 2023-05-02:
Did you run the latter with sanitize too? Seems there is some dangling pointer in the last line. I can have a look at it too

tudortimi, 2023-05-02:
@wallento I ran them both with the same `verilator`, compiled with `-fsanitize=address,undefined`.

If you want to reproduce the fail, it's on https://github.com/svunit/svunit/pull/233

To reproduce (assumes there is an executable called `verilator` on `PATH`):

```
source Setup.<your-shell>
cd test/sim_3
runSVUnit -c_arg --debug -c_arg --gdbbt
```

wsnyder, 2023-05-03:
Rereading the code based on your example, try changing

       || !prevp->filep()->sameTree(nodep->filep())) 

to

      || (prevp->filep() && nodep->filep() && !prevp->filep()->sameTree(nodep->filep())))
      
and if that works (based on sanitizer passing), please consider submitting a pull request to change it, thansk.

tudortimi, 2023-05-03:
Also, do you get the feeling that the sanitizer related fail and this other error are two unrelated issues?

wsnyder, 2023-05-03:
The super.new is different, let's please use this issue to close the display one. and please make a new one for super.new.

tudortimi, 2023-05-03:
I kept this and opened a new one, since this one had most of the context for the `super.new` issue and I had already references to it.

tudortimi, 2023-05-03:
I did some reading of the internals docs and some digging through the tree files and I noticed the following:

```
    1:2:2:3:1: NEW (PZ) <e15144> {o95aj} @dt=0@  new pkg=(MU) -> FUNC (AV) <e20815#> {n56ax} @dt=(FB)@(w0)void  new [METHOD]
```

For the `super.new` call, the statement doesn't have a data type set (hence the error message). I compared this to other `super.new` calls and there the data types are set:

```
    1:2:2:3:1: NEW (GIB) <e25095#> {p72aj} @dt=(FB)@(w0)void  new pkg=(MU) -> FUNC (AV) <e20815#> {n56ax} @dt=(FB)@(w0)void  new [METHOD]
```

wallento, 2023-05-04:
It would be interesting to isolate the problematic call. I would suggest to first look for the file and line information and then isolate the problem, this might shorten the debug time.

wallento, 2023-05-04:
Ah, I see it comes from the library, so probably not too much value there..

tudortimi, 2023-05-04:
I usually try to progressively remove code to end up with a minimal example. Since Verilator generally takes this code, I'm not so sure of my chances, because I assume the problem isn't just some one little thing, but a combination of many.

tudortimi, 2023-05-07:
I debugged a bit more. I have some code like the following:

```
// file svunit_pkg.sv
package svunit_pkg;
  `include "svunit_base.svh"
  `include "svunit_testcase.svh"
endpackage

// file svunit_base.svh
class svunit_base;
  function new(string name);
    // ...
  endfunction
endclass

// file svunit_testcase.svh
class svunit_testcase extends svunit_base;
  function new(string name);
    super.new(name);
  endfunction
endclass
```

`V3Width` visits the node for `super.new()` before the node for `suvnit_base::new`. It doesn't see any width for the `super.new` node and tries to do `dtypeFrom` on the node for `svunit_base::new`. The node for `svunit_base::new` doesn't have any width either, so it propagates "no width". Should the node for `svunit_base::new` have been visited first?

tudortimi, 2023-05-07:
I've managed to strip the code down to an absolute minimum while still reproducing the issue.: [compile-crash-due-to-dtype.zip](https://github.com/verilator/verilator/files/11415880/compile-crash-due-to-dtype.zip)

wsnyder, 2023-05-07:
That's an empty archive I believe.

tudortimi, 2023-05-07:
I reuploaded. In any case, I squeezed everything even further into one file:

```
class svunit_base;
  function new(string name);
  endfunction
endclass

class svunit_testcase extends svunit_base;
  function new(string name);
    super.new(name);
  endfunction
endclass
  

module dut_unit_test;
  svunit_testcase svunit_ut = new("dut_ut");
endmodule

module testsuite;
  dut_unit_test dut_ut();
endmodule

module testrunner();
  testsuite ts();
endmodule
```

wsnyder, 2023-05-07:
(Side note, it would be super cool if Verilator could put the content of all files it read into one single file like this, to aid in creating cases.)

With --debug it does make .vpp files, but not one.  Mind filing an issue suggesting that?

tudortimi, 2023-05-07:
Actually, it seems that `testrunner` is not necessary. I've updated the code.

wsnyder, 2023-05-07:
Ok, your analysis seems correct it needs to iterate into the function definition.  Thanks for the debug, working on a patch.

tudortimi, 2023-05-07:
I'm just sad I didn't work on creating the example before going down the rabbit hole of print-debugging through the Verialtor code :rofl:

closed 2023-05-07 by wsnyder