verilator/verilator #4152: Signals updated at wrong times when using verilator.

phelter, 2023-04-28:
Using a third-party processor that is from a company that is also an upper body appendage, so cannot provide you a code drop.

Version: Verilator 5.006 2023-01-22 rev v5.006 on Ubuntu-22.04

When running a simulation of the code - a portion of the code is not operating as expected.

First note there are several warnings that are left ignored since this is third-party code.
For this specific code instance where the issue occurs the following warnings are turned off:
`TIMESCALEMOD`, `UNOPTFLAT`, and `UNUSEDSIGNAL` via an included set of `waiver.vlt` files in the file-list which contains:

waiver.vlt eg:
```txt
`verilator_config
lint_off -rule TIMESCALEMOD -file "*<dir>/<filename>.v"
....
etc.
```

The verilator command is as follows:

```bash
/usr/local/bin/verilator_bin --compiler clang --prefix Vtop_core --Mdir Vtop_core.dir --make cmake -Wall -Wpedantic --assert -sv --hierarchical --trace --trace-structs --coverage -DVERILATOR_WORKAROUND=1 --no-timing --top top_core --cc \
-I <root>/...  -I<root>/... etc.  \ # A bunch of these 
<File list - lots of them mix of system verilog and verilog> \
<root>/../waiver.vlt \ # Several of these for various modules - these include lint_off -rule <RULE> -file "*dir/filename.v"
```

It all compiles well and I have not had this issue with any of the designs I've done (although they've been much smaller), that are warning clean - i.e. no warnings to mention.

After compilation I have a C++ testbench environment that uses GoogleTest as the Test Framwork and some UVM like drivers (but written in C++)  that generate the stimulus.  The biggest thing is it doesn't work on every tick but identifies the external minimum tick and updates on the next required change - i.e. time advances multiple ns depending on the inputs driven and when they need to be updated.

That being said this approach has worked for me on several designs so far but is failing me now with this particular verilog code.

there is a flop which is defined as:

```verilog
  always @ (posedge hclk or negedge hreset_n)
    if (!hreset_n)
      foo <= {32{1'b0}};
    else if (foo_we)
      foo <= nxt_foo;
```

note nxt_foo is generated via a large logic and/or tree eg:
```verilog
always @(... list of inputs)
  case(bar_state_i)
   `BAR_STATE_1 : begin
       nxt_foo = 32'hFFFFFFFF;
       foo_we  = ready;
   end
   `BAR_STATE_2, `BAR_STATE_3, BAR_STATE_4 : begin
      nxt_foo = {some_value[31:2], 2'b00};
      foo_we = ready & some_enable;
   end
   .... Several more of these.
   `default : begin
     nxt_foo = {32{1'bx}};
     foo_we = 1'bx;
  endcase
```

And under certain circumstances the waveforms show that on some `posedge hclk` scenarios (not all), the value of `nxt_foo` is updated before the value of foo is captured, meaning for some clock cycles `foo` and `nxt_foo` change to the same value (i.e. what `nxt_foo` updated to).

Sorry for the rudimentary diagram but thought this was simpler than wavedrom decipher if you don't have a viewer.
So at cycle 3 nxt_foo = 0xAAAAAAAA, but that doesn't get updated - cycle 4's value gets updated on both nxt_foo and foo.

I've added what is expected below but that isn't what the verilated C code is generating.

```txt
Cycle            |    0    |    1    |    2    |    3    |    4    |    5    |    6    |
hclk         ____|----|____|----|____|----|____|----|____|----|____|----|____|----|____|----|____
foo_we    _________________|---------|_________|-------------------|____________________
nxt_foo   _________________Xffffffff X_________Xaaaaaaaa X00000123 X____________________
Actual:
foo       ___________________________Xffffffff           X00000123
Expected:
foo       ___________________________Xffffffff           Xaaaaaaaa X00000123
```

How would I go and investigate this bug? - I looked at the verilated code and might disable code coverage to remove all of the unruly stuff that adds - and look at what gets updated first and why - any other options?

Also note with another testbench I have run through this same scenario with Questa Sim (New name for Mentor Graphics) - and it does show the Expected values and differs from what Verilator produces.

phelter, 2023-04-28:
A few test runs (Note each one was additive on the previous trial):
* Removing `--coverage` - no change (still shows issue) but much faster (5x) to run.
* Fixed the particular Warning.UNOPTFLAT within the file in question  - no change
* Removing `--hierarchical` - no change (small speed increase)
* Added in verilate command `--clk hclk` - no change
* Removed fix for UNOPTFLAT - now using `split_var` in configuration file. - no change
  * Note there are areas of the code where I just disabled linting:
  * Above affected area: CMPCONST, IMPLICIT, PINCONNECTEMPTY, PINMISSING, EOFNEWLINE, TIMESCALEMOD, UNDRIVEN, UNSIGNED, UNUSEDPARAM, UNUSEDSIGNAL
  * In Memory Model attached to processor: (DECLFILENAME, IMPLICITSTATIC, WIDTH, ASSIGNDLY, BLKSEQ, COMBDLY, LATCh, STMTDLY, SYNCASYNCNET, UNDRIVEN, UNUSEDPARAM, UNUSEDSIGNAL
  * In processor: COMBDLY, DECLFILENAME, SYNCASYNCNET, PINCONNECTEMPTY, TIMESCALEMOD, `split_var`, UNUSEDPARAM, UNUSEDSIGNAL, VARHIDDEN, WIDTH
* Added back in `--hierarchical` and added several `hier_block -module ""` into the config to try and isolate the code - no change to generated code - did not separate out the hierarchy there - same file set as with just --hierarchy (see below).
* Added `no_inline -block "module"` into the config to try and isolate the code - verilator fails without a message. - cannot do this for some reason.

So despite not being able to isolate the specific code - used what was separated and looked for the net names within the verilated code:

I searched in the verilated code where all of the instances of the regexp `(_|nxt)_foo ` are updated `assigned` and added either:

```cpp
   std::cout << "foo: " << __func__ << ":" << __LINE__ << " value: " << vlSelf->...._foo << std::endl;
   // OR
   std::cout << "nxt_foo: " << __func__ << ":" << __LINE__ << " value: " << vlSelf->...._nxt_foo << std::endl;
```

and Re-ran - ignoring the trace ones.

Oddly enough the assignments of these were split among multiple files: - where `...` is `<module>___024root__DepSet`
* V..._h90f927d6__0.cpp - in function `V...___ico_sequent_TOP_0()` - only `nxt_foo`
* V..._h90f927d6__1.cpp - in function `V...___act_comb__TOP_2()` - only `nxt_foo`
* V..._h90f927d6__2__Slow.cpp - in function `V...___stl_sequent__TOP_2()` - only `foo` - consumed.
* V..._h90f927d6__4__Slow.cpp - in function `V...__-stl_sequent__TOP__4()` - only `nxt_foo`

* V.._hd7c55739__0__Slow.cpp - in function `V...___ctor_var_reset()` - both `foo` and `nxt_foo` -reset.
* V..._hd7c55739__0.cpp - in function `V...___nba_sequent__TOP__4` - only `foo`
* V..._hd7c55739__3.cpp - in function `V...___nba_comb__TOP__45` - only `foo` - consumed.
* V..._hd7c55739__4.cpp - in function `V...___nba_comb__TOP__143` - only `nxt_foo` - strange to have twice?
* V..._Trace__0.cpp - only `foo`
* V..._Trace__1__Slow.cpp - only `foo`

The order of function calls - Ignoring the startup and setup - and getting to a reset and then start of the execution:
```txt
Line  Info    <-- Notes.
----  ----------------------------------------------
 233649 Hard Reset
 233650 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233651 nxt_foo: Vmd_top_core___024root___nba_comb__TOP__143:9022 value: 4
 233652 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233653 nxt_foo: Vmd_top_core___024root___act_comb__TOP__2:14851 value: 4
 233654 foo: Vmd_top_core___024root___nba_sequent__TOP__4:3016 value: 0
 233655 nxt_foo: Vmd_top_core___024root___nba_comb__TOP__143:9022 value: 4
 233656 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233657 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233658 Hard Reset Done
 233659 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233660 nxt_foo: Vmd_top_core___024root___act_comb__TOP__2:14851 value: 4
 233661 foo: Vmd_top_core___024root___nba_sequent__TOP__4:3016 value: 0            <-- Reset released.
 233662 nxt_foo: Vmd_top_core___024root___nba_comb__TOP__143:9022 value: 4
 233663 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233664 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233665 nxt_foo: Vmd_top_core___024root___act_comb__TOP__2:14851 value: 4
 233666 foo: Vmd_top_core___024root___nba_sequent__TOP__4:3016 value: 0
 233667 nxt_foo: Vmd_top_core___024root___nba_comb__TOP__143:9022 value: 4
 233668 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233669 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233670 nxt_foo: Vmd_top_core___024root___act_comb__TOP__2:14851 value: 4
 233671 foo: Vmd_top_core___024root___nba_sequent__TOP__4:3016 value: 0
 233672 nxt_foo: Vmd_top_core___024root___nba_comb__TOP__143:9022 value: 4
 233673 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233674 nxt_foo: Vmd_top_core___024root___ico_sequent__TOP__0:11730 value: 4
 233675 nxt_foo: Vmd_top_core___024root___act_comb__TOP__2:14851 value: 4
 233676 foo: Vmd_top_core___024root___nba_sequent__TOP__4:3016 value: 0
[... 58 lines cut ...]
```cpp
    // inside the constructor - m_context(std::make_unique<VerilatedContext>())
    //         m_dut = std::make_unique<T>(m_context.get(), "TOP");

    inline void traceEval()
    {
        m_dut->eval();
        if (m_trace)
        {
            m_trace->dump(m_context->time());
        }
    }

    template<class Rep, class Period = std::ratio<1>>
    inline void timeIncr(const std::chrono::duration<Rep, Period> &duration)
    {
        if (duration == duration.zero())
        {
            throw std::runtime_error("Cannot increment time by zero.");
        }
        m_context->timeInc(std::chrono::duration_cast<Duration_t>(duration).count());
        traceEval();
    }

```

wsnyder, 2023-04-28:
Did you try --timing?

I would avoid --hierarchical entirely until this is debugged.

See what posedge blocks or inputs all the clocks come from.  A common problem is a signal comes from multiple clock domains and a race is exposed.  (Older Verilators also had scheduling problems with multiple clock gaters but this is fixed.)

It's a bit complicated to describe, but since you seem able to dive in, please but see if the ordering graph you get from --dump-graph in ..._072_nba_orderg_done.dot involving the signals shows a graph edge between the proper signals.

phelter, 2023-04-28:
This is single clock domain block (hclk).
Note: - the verilog models for my memories use #0 - ZERODLY so had to comment those out.  I've originally been running with --no-timing.

With `--timing` and removing `--hierarchical` - no change.

```bash
usr/local/bin/verilator_bin --compiler clang --prefix Vmd_top_core --Mdir <build>/Vmd_top_core.dir --make cmake -Wall -Wpedantic --assert -sv --trace --timing --trace-structs --clk hclk --top md_top_core --cc -I <All include dirs> <all verilog system verilog files>
```

One other thing I forgot to mention is all of this is with C/C++ running with Debug mode in CMake - running Clang 14.0.

The input clock comes from my testbench C++ driver modifying the value of the `m_dut->hclk` value at specific intervals.  I do not rely on a testbench layer around the device-under-test (dut) for driving signals, I drive them directly from the C++ code.

My number is `..._083_nba_orderg_done.dot`  - assuming you wanted the nba_orderq_done.dot over other stuff.

I'm generating the pdf now - taking a while.

wsnyder, 2023-04-29:
You probably do not want tomale a pdf, ot will be too large.

Do any of the signals feeding the broken signal come from inputs to the model?  Try to drive nonclocks on negedge to prevent races.

phelter, 2023-04-29:
No no inputs to the model drive these signals - not even the hclk or rstb do - they go through a level of clock gates or synchronizers beforehand.

This is nested well within the model - the only thing that does drive these is the sram model.  I'm going to swap that out for one that mimics it instead of the behavioral model they provided and see if that fixes things.  The sram model is a timing model for simulation with timing etc. So it may be the culprit.

wsnyder, 2023-04-29:
Removing the #0's may have caused the problems, good idea to try another model.

phelter, 2023-04-30:
Ok - so it was the Memory model generated by the memory compiler that caused the issue.  When I changed that to a simpler but equivalent behavioral model, the signals are operating as expected.  Also making the memory model just a behavioral model with a much simpler implementation (no `'celldefine` and all of the setup/hold etc. made the test run time go from 68s down to 8s.  So win-win.

Thanks for the assistance.  Will ensure in the future I generate behavioral models for the memories before I go about trying to simulate.  The ones that I was using are generated are for a simulator that supports both gate-level and rtl level verilog.

wsnyder, 2023-04-30:
Glad it's solved.  Most likely your normal simulator will also speed up with behavioral models BTW.

closed 2023-04-30 by wsnyder