verilator/verilator #4154: invalid initialization error during 'make test'

sia84, 2023-05-01:
make test produces following error:

[ 41%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop.cpp.o
[ 41%] Building CXX object CMakeFiles/example.dir/opt/verilator/verilator-5.010/examples/make_hello_c/sim_main.cpp.o
[ 50%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp.o
[ 50%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__Slow.cpp.o
[ 50%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_heccd7ead__0.cpp.o
[ 58%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_heccd7ead__0__Slow.cpp.o
[ 66%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop__Syms.cpp.o
[ 66%] Building CXX object CMakeFiles/example.dir/opt/verilator/verilator-5.010/include/verilated.cpp.o
[ 75%] Building CXX object CMakeFiles/example.dir/opt/verilator/verilator-5.010/include/verilated_threads.cpp.o
/opt/verilator/verilator-5.010/include/verilated.cpp: In constructor ‘VerilatedModel::VerilatedModel(VerilatedContext&)’:
/opt/verilator/verilator-5.010/include/verilated.cpp:3012:24: error: invalid initialization of non-const reference of type ‘VerilatedContext&’ from an rvalue of type ‘<brace-enclosed initializer list>’
     : m_context{context} {}
                        ^
gmake[4]: *** [CMakeFiles/example.dir/opt/verilator/verilator-5.010/include/verilated.cpp.o] Error 1
gmake[4]: Leaving directory `/opt/verilator/verilator-5.010/examples/cmake_hello_c/build'
gmake[3]: *** [CMakeFiles/example.dir/all] Error 2
gmake[3]: Leaving directory `/opt/verilator/verilator-5.010/examples/cmake_hello_c/build'
gmake[2]: *** [all] Error 2
gmake[2]: Leaving directory `/opt/verilator/verilator-5.010/examples/cmake_hello_c/build'
make[1]: *** [run] Error 2
make[1]: Leaving directory `/opt/verilator/verilator-5.010/examples/cmake_hello_c'
make: *** [examples] Error 10

I am using gcc 8.2.0 on CentOS 7.9. Any ideas why ?

wsnyder, 2023-05-01:
Looks like a compiler bug, try changing the code to ": m_context(context) {}"

sia84, 2023-05-01:
@wsnyder Thanks! that got me past that error. Now I am getting another error:

[ 23%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_heccd7ead__0.cpp.o
[ 23%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0.cpp.o
[ 35%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop.cpp.o
[ 41%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_h84412442__0__Slow.cpp.o
[ 41%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__DepSet_heccd7ead__0__Slow.cpp.o
[ 47%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop__Trace__0.cpp.o
[ 52%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop___024root__Slow.cpp.o
[ 52%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop__Trace__0__Slow.cpp.o
[ 58%] Building CXX object CMakeFiles/example.dir/CMakeFiles/example.dir/Vtop.dir/Vtop__Syms.cpp.o
/opt/verilator/verilator-master/examples/cmake_tracing_c/build/CMakeFiles/example.dir/Vtop.dir/Vtop.cpp: In constructor ‘Vtop::Vtop(VerilatedContext*, const char*)’:
/opt/verilator/verilator-master/examples/cmake_tracing_c/build/CMakeFiles/example.dir/Vtop.dir/Vtop.cpp:22:28: error: invalid initialization of non-const reference of type ‘VlWide<3ul>&’ from an rvalue of type ‘<brace-enclosed initializer list>’
     , rootp{&(vlSymsp->TOP)}
                            ^
I also tried gcc 10.2.0 and gcc 11.2.0 and got the same error. What's the recommended gcc version ?

wsnyder, 2023-05-02:
Ubuntu 20.04 and 22.04 with gcc 9.3 and 11.3 are known to work.  I'd think gcc 11.2 would work; is there something else different about your compiler, did you set some special compiler flags?

The error you posted doesn't make sense, it looks like part of the line was truncated?

closed 2023-05-02 by sia84