verilator/verilator #4156: SVA assertion false positive when used in a sub-module

quentinschibler, 2023-05-01:
Assertions in sub-modules sometimes trigger when they should not.
I have a simple register module:
```
module register
#(
	parameter WORD_WIDTH = 8,
	parameter RESET_VALUE = 0
)
(
	input  logic clk,
	input  logic rst_n,
	input  logic en,
	input  logic [WORD_WIDTH-1:0] word_in,
	output logic [WORD_WIDTH-1:0] word_out
);

	always @(posedge clk)
	begin
		if(en) word_out <= word_in;
		if(!rst_n) word_out <= RESET_VALUE;
	end

`ifdef ASSERT
	always @(posedge clk)
	begin
		if($time > 1)
		begin
			assert($past(rst_n) || word_out == RESET_VALUE);
			assert($past(en || !rst_n) || $stable(word_out));
			assert
			(
				$past(!en || !rst_n) || 
				($past(word_in) == word_out)
			);
		end
	end
`endif
```
If do a simulation with random inputs (2^32 steps), none of the assertions are triggered. In particular, this does not fail :
```
0 TOP.register:
	rst_n: 1, en: 1, word_in: 131, word_out: 222,  
1 TOP.register:
	rst_n: 1, en: 0, word_in: 216, word_out: 131,  
2 TOP.register:
	rst_n: 1, en: 1, word_in: 165, word_out: 131,
```
If I instantiate one of this register in a top module, and simulate that, one assertion fail despite being given
the exact same inputs:

```
0 TOP.skid.word_out_reg:
	rst_n: 1, en: 1, word_in: 131, word_out: 17,
1 TOP.skid.word_out_reg:
	rst_n: 1, en: 0, word_in: 216, word_out: 131,
2 TOP.skid.word_out_reg:
	rst_n: 1, en: 1, word_in: 165, word_out: 131,
[2] %Error: base_register.sv:29: Assertion failed in TOP.skid.word_out_reg: 'assert' failed.
%Error: /home/quentin/dev/cord/src/base_register.sv:29: Verilog $stop
 ```

Also not entirely related, but is it possible to enable assertions only for the top module ? It would solve this issue temporarily for me, and it would speed up simulation time.

Using Verilator 5.008 2023-03-04 rev UNKNOWN.REV, on linux

wsnyder, 2023-05-02:
Maybe suspect the variable behind implementing $past is getting randomly initialized differently?  Perhaps you want to start assertions well past a few clocks have happened, e.g. $time > 10 or something?

quentinschibler, 2023-05-02:
I tried with `$time > 10`, it didn't change anything. I would have been surprised anyway since the assertion is triggered at the 3rd clock cycle, not the first.

quentinschibler, 2023-05-02:
I was just being careless and updated my inputs on both edge of the clock. I fixed that and it is now working. I still don't really understand why the assert didn't triggered on the first example though

closed 2023-05-02 by wsnyder