verilator/verilator #4162: Verilator's $fopen reads all zeros from other directories

RadicalRoman, 2023-05-02:
If you call the $fopen task from a directory that doesn't contain the file you're opening, Verilator reads all zeros.

wsnyder, 2023-05-03:
-y only affects the compilation, so that won't matter.  

Does e.g. '$system("pwd") indicate you are running from where you expect (assuming you are passing a relative filename)?

Is the $fopen return value incorrect?  Please suggest a change to test_regress/t/t_sys_file.v (or make a new similar test) that shows the problem by passing on another simulator and failing on Verilator.

RadicalRoman, 2023-05-03:
Regardless of the directory from which I run the testbench, $system("pwd") yields 0.

Sounds good! I can dig into this and get back to you. Thanks!

wsnyder, 2023-05-03:
Sorry, I meant make sure the directory that the system call prints on the screen is what you expected.

That is if it prints "foo" you expected to be in foo and so $fopen("bar") will thus be expected to look in "foo/bar".

closed 2023-05-27 by wsnyder