verilator/verilator #4168: No Warning on when Omitting Base from Verilog Integer Literal

sifferman, 2023-05-03:
Hello, I noticed that Verilator does not warn when a SystemVerilog unsized literal (`'0`, `'1` `'x`, `'z`) is used in a Verilog file. (Vivado will correctly give an error in this case)

For example, no error/warning is given on:

```verilog
wire [7:0] a = '0;
wire [7:0] b = '1;
wire [7:0] c = 'x;
wire [7:0] d = 'z;
```

Lint commond
```bash
verilator --lint-only -Wall --language 1364-2005 <file.v>
```

Verilator version
```plain
Verilator 5.007 devel rev v5.006-7-g4a8cfe367
```

I can try to solve this on my own, but I have not looked much at the Verilator source code. Could someone direct me to which file(s) I should be editing?

wsnyder, 2023-05-03:
Verilator doesn't generally make any effort to check code for a newer version of the standard should be flagged as an error when using an older version of the standard.  However, if you want to add this check, feel free.  Make a test_regress/t/t_number_v_bad.v with the code (under a module), then make a .pl and .out file similar to t_number_bad.pl/.out.  Then in  V3LinkParse you can "visit AstConst" to print an error if it is an (constp->num().autoExtend() and nodep->fileline()->language() < V3LangCode::L1800_2005.  (See similar "parameter requires default value, or use IEEE 1800-2009 or later") message.

sifferman, 2023-05-04:
Oh, neverminded, I misread your response.

All good now!

wsnyder, 2023-05-06:
Thanks for the good pull.

PR #4172 (2023-05-06): Added NEWERSTD warning (#4168)
Fixes #4168
closed 2023-05-06 by wsnyder