verilator/verilator #4173: Nested unpacked struct causing segfault

Risto97, 2023-05-06:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```
module test2;
    typedef struct {
        struct {
            struct {
                logic [31:0] next;
            } val;
        } el[1];
    } field_combo_t;
    field_combo_t field_combo;
endmodule
```

`/tools/work/verilator_compile/5.10/bin/verilator_bin --build --cc ../test2.sv`
```
Segmentation fault (core dumped)
```

What 'verilator --version' are you using?  Did you try it with the git master version?
`Verilator 5.011 devel rev v5.010-25-g76f5de6e5`
But I tried also 5.006 tag

May we assist you in trying to fix this in Verilator yourself?

Yes I can give it a try if I get some directions.

wsnyder, 2023-05-07:
It's confused by a combination of missing/implied typedefs, unpacked, and some internal stuff. Adding typedefs (and not using --unpacked) should work around it, I'll get to a fix.

```
   typedef struct {
      logic [31:0] next;
   } val_t;
   typedef struct  {
      val_t val;
   } el_t;
   typedef struct  {
      el_t el[1];
   } str_t;

   str_t str;
```

closed 2023-05-07 by wsnyder

Risto97, 2023-05-07:
Great, thanks a lot, that was fast.
I tested it with a larger design using multidimensional arrays of struct and such and it works now.

I did not intend to claim the credit for the fix, but thank you.