verilator/verilator #4175: if i want to waive all report  about folder(errors or wraning),what should i write to my config.vlt?

ZhenzongGu, 2023-05-07:
this doesn't work.
[image]

wsnyder, 2023-05-07:
To help process this, and search, please convert screen shots to text.

    lint_off -file "*/mem_wrap/.../*" 

Anyhow this works fine for me, you didn't provide your verilator command line - perhaps you aren't putting this file first, you need to, I'll presume that is the problem.  If you are putting it first, please provide a complete self contained example showing the problem and I'll reopen.

closed 2023-05-07 by wsnyder

ZhenzongGu, 2023-05-08:
> To help process this, and search, please convert screen shots to text.
>
> ```
> lint_off -file "*/mem_wrap/.../*"
> ```
>
> Anyhow this works fine for me, you didn't provide your verilator command line - [...]

this is my my verilog,
[image]
this is my waive.vlt
[image]
this is command 
[image]
this is report 
[image]
this is `pwd`
[image]
I don't unstand why it still report Warning?Can you tell please?thank you .

wsnyder, 2023-05-08:
The filename is foo.v which does not match *verilator_test*.   If you want the filename to have the current directory, pass `pwd`/foo.v.

ZhenzongGu, 2023-05-08:
> The filename is foo.v which does not match _verilator_test_. If you want the filename to have the current directory, pass `pwd`/foo.v.

If I want to ignore all files under veriloator_test folder ,Should I do ?

ZhenzongGu, 2023-05-08:
> 

If there are a dozen files under a folder,Can I use regular expressions to mask entire directories,instead of listing the names of eachfile.

ZhenzongGu, 2023-05-08:
> Pass
> 
> ```
>     `pwd`/foo.v`
> ```
> 
> on the command line, and use the waiver file as you described.

If there are a dozen files under a folder,Can I use regular expressions to mask entire directories,instead of listing the names of eachfile.

wallento, 2023-05-08:
Yes, * wildcard should work

ZhenzongGu, 2023-05-08:
> Yes, * wildcard should work

But It doesn't work.Did I write it in the wrong format?
[image]