verilator/verilator #4178: Add support for nested classes

tudortimi, 2023-05-07:
Is support for nested classes something reasonably easy to add (e.g. for a beginner)?

tudortimi, 2023-05-07:
I generally use nested classes a lot to avoid "polluting" scopes. Sometimes, they also make more sense, because the nesting class creates a nice sub-scope.

wsnyder, 2023-05-07:
I presume we will flatten out the classes, that is a nested class will become non-nested in the C++, otherwise things will get "interesting". So, the first part is to support elaboration of any nest, then supporting nesting where there are class parameters (inside/outside/both).  The hard part will be stomping out the bugs that result.

Likely a few week project, if you'd like to start.  We could always e.g. support nesting only when there are no parameters, then improve from there.

tudortimi, 2023-05-07:
Sure, how would one start with just the "no parameters" part?

wsnyder, 2023-05-07:
First, make a bunch of tests that try some complicated things and make sure they pass on other simulators.  I'd suggest separate out non-parameter and parameter tests.  Some of the complicated things to try in both the inner/outer:

  parameters
  extends
  function new
  member variables
  member functions
  static methods
  virtual methods
  accessing outer static/virtual/member functions from inner

That is, similar to what much of what the current tests check, but in the inner context.

This can be a pull we merge without any other support ready, the tests would just check Verilator will error out. 

Then start on support.

Delete the current error, run with --debug --debugi-V3LinkDot to see how the .tree files look, and the .txt file that describes the symbol table.  Get through V3LinkDot stage (this will be probably 70% of the work).

V3Class (I think) needs to learn how to pull apart the nested classes to flatten them.  Past that point (once flat) in the processing pipeline, it should be just like there's no nested classes to the internals, so hopefully only a few changes may result.

tudortimi, 2023-05-09:
Understood, I'll start thinking of tests.

tudortimi, 2023-09-18:
Can you assign this to me?

wsnyder, 2023-09-18:
Great, we welcome work on this.

In addition to suggestions above, you should audit all "visit (AstClass" to see if what they do makes sense if they are recursed upon due to a class-in-class.  Generally they should be using "VL_RESTORER" so they handle it, but likely there's some escapes as this hasn't been tested.

jklapel, 2024-07-09:
Has any progress been made on this issue?  We are testing out Verilator and use nested classes fairly extensively in our testbench framework.

wsnyder, 2024-07-09:
Not to my knowledge, and as nested classes are not used in the UVM core, and this is not trivial, this is unlikely to get a lot of attention. So please consider making a pull if you want it.

closed 2025-02-17 by wsnyder