verilator/verilator #4180: Multidriven warning with both drivers being the same line

TimK-22, 2023-05-09:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
[t_lint_multidriven_same.tgz](https://github.com/verilator/verilator/files/11428080/t_lint_multidriven_same.tgz)

What 'verilator --version' are you using?  Did you try it with the git master version?
5.010, not with the master

What OS and distribution are you using?
Rocky 9 

-------------

I spent some time debugging this issue and found it was due to incorrect AST -> DFG generation.   What seems to happen is that in the AST the CONCAT 
```
    1:2: ASSIGNW 0x131b120 <e1788#> {e58ai} @dt=0x125fdf0@(G/w6)
    1:2:1: VARREF 0x131af00 <e1785#> {e58ai} @dt=0x125fdf0@(G/w6)  __Vcellout__dut__data_out [RV] <- VAR 0x131ad80 <e1778#> {e72ax} @dt=0x125fdf0@(G/w6)  __Vcellout__dut__data_out MODULETEMP
    1:2:2: CONCAT 0x1309d40 <e1786#> {e59bb} @dt=0x125fdf0@(G/w6)
    1:2:2:1: VARREF 0x125f560 <e1617> {e59at} @dt=0x1302cd0@(G/w2)  localval [LV] => VAR 0x1304a90 <e1600> {e29aq} @dt=0x1302cd0@(G/w2)  localval [VSTATIC]  VAR
    1:2:2:2: VARXREF 0x125f670 <e1618> {e60ba} @dt=0x1301fb0@(G/w4)  value [LV] => .=test_i VAR 0x12800e0 <e1619> {e8aq} @dt=0x1301fb0@(G/w4)  value [VSTATIC]  VAR
  ```
is converted to a `DfgSel` but the presence of the `VARXREF` means `V3DfgAstToDfg::convertAssignment` returns false as it can't convert it.  This means the AST is not stripped from the tree and remains.  But the `DfgSel` remains and is eventually converted back to AST.  Now the AST as both the code above and the code below.

```
    1:2: ASSIGNW 0x131e1a0 <e1997#> {e58ai} @dt=0x1302cd0@(G/w2)
    1:2:1: SEL 0x131def0 <e1994#> {e59at} @dt=0x1302cd0@(G/w2)
    1:2:1:1: VARREF 0x131da20 <e1983#> {e72ax} @dt=0x125fdf0@(G/w6)  __Vcellout__dut__data_out [RV] <- VAR 0x131ad80 <e1778> {e72ax} u3=0x1 @dt=0x125fdf0@(G/w6)  __Vcellout__dut__data_out MODULETEMP
    1:2:1:2: CONST 0x131db30 <e1984#> {e59at} @dt=0x1307a90@(G/w32)  32'h4
    1:2:1:3: CONST 0x131dd10 <e1985#> {e59at} @dt=0x1307a90@(G/w32)  32'h2
    1:2:2: VARREF 0x131e090 <e1995#> {e29aq} @dt=0x1302cd0@(G/w2)  localval [LV] => VAR 0x1304a90 <e1600> {e29aq} u3=0x1 @dt=0x1302cd0@(G/w2)  localval [VSTATIC]  VAR
 ```
 In the next DFG conversion this gets flagged as multidriven with the filelines exactly the same.
 
 In testing, I found this was:
 - When the variables in the concat were interfaces
 - When the non-interface variable came before the interface variables.  If the interface variable came first then the DfgSel seems not to have been produced at all.  We have switched orders as a workaround.

Tim

wsnyder, 2023-05-10:
Presumably we need to remove the DfgSel?  @gezalore?

Would you be willing to attempt a pull to fix that?

wsnyder, 2025-09-09:
Master no longer reports this as multidriven.  Reading the comments it appears there might have also been a wrong result involved, if so please file a new issue with self-checking test, thanks.

closed 2025-09-09 by wsnyder