verilator/verilator #4181: Segfault / Broken link when enum parameter is implicitly cast to integer

TimK-22, 2023-05-09:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

[t_enum_param_segfault.tgz](https://github.com/verilator/verilator/files/11428150/t_enum_param_segfault.tgz)

What 'verilator --version' are you using?  Did you try it with the git master version?
5.010, no
What OS and distribution are you using?
Rocky 9

--------------

This problem occurs either as a segfault or as a broken link (it seemed to depend whether I ran debug or not).   From debugging in gdb the issue seemed to be that a pointer was pointing at an `AstEnum` class that had been deleted while running the `WidthVisitor`.  It was mostly there but the vtable was corrupted (gdb thought it was a plain AstNode).   

We worked around this by ensuring an explicit cast.  This seemed to fix the issue.

Tim

wsnyder, 2023-05-10:
Looks like V3Dead thinks the enum is unused (since the enum type isn't referenced).  An enum value alone needs to mark it used.

Perhaps you'd be willing to make a pull request to fix that?

wsnyder, 2025-01-04:
Fixed with #5714

closed 2025-01-04 by wsnyder