verilator/verilator #4183: `release` statements affect combo logic incorrectly

kbieganski, 2023-05-09:
Verilator revision befb415f278af51fc3ecc3dda1dfe820de023bd1, but the bug affects all v5 versions and some earlier ones.

Test case:
```systemverilog
// verilator lint_off MULTIDRIVEN
module t(input clk);
  initial begin
    $dumpfile("dump.vcd");
    $dumpvars;
  end

  logic [31:0] lhs1, lhs2, rhs;
  logic cond = 0;

  always_comb lhs1 = rhs;
  assign lhs2 = rhs;

  always @(posedge clk) rhs = '1;

  always @(negedge clk) begin
    if (!cond) begin
      release lhs1;
      release lhs2;
    end
  end
endmodule
```

Waveform:

[image]

`lhs1` and `lhs2` are oscillating for some reason, when they should equal `rhs`.

The `always_comb` bug also appears on v4.228.

[image]

This seems to be causing a regression in Calliptra: https://github.com/chipsalliance/caliptra-rtl/issues/79

wsnyder, 2023-05-10:
Thanks for the good writeup. Force/releases weren't designed to be ordered in clocked blocks along with the "original signals"; they introduce new variables that are decoupled from the original variables.  So this isn't surprising.   I'm not sure what the fix would be.

kbieganski, 2023-05-11:
Maybe something like [this](https://github.com/antmicro/verilator-1/commit/ea60fd9da46624adeed65b0af5d499a328acb422)? On release, instead of always assigning the forced val, only do it if force enable was actually set. Doesn't break any regression tests, fixes this test case and fixes the issue in Caliptra.

I was going to submit a PR, but ran into another bug: #4190

wsnyder, 2023-05-11:
That approach seems reasonable.  But should it be a bit-by-bit replacement (e.g. "A & sel | B & ~sel") instead of a whole-signal conditional ("|sel ? A : B")?

kbieganski, 2023-05-11:
Ok, from reading the code I think your suggestion is correct.

PR #4192 (2023-05-12): Assign forced val on forced assign release conditionally (#4183)
This patch concerns forced assignments that override non-continuous assignments.
Forced val is only assigned on release if it was forced in the first place.
Fixes #4183.