verilator/verilator #4184: Support for Immediate assert

phelter, 2023-05-09:
Would like to have support for immediate asserts - see: https://www.doulos.com/knowhow/systemverilog/systemverilog-tutorials/systemverilog-assertions-tutorial/

```systemverilog
assert (ADDR_SIZE >= 8) else $error("Must be 1024 KB minimum address size");
```

I've currently worked around this by using:

```systemverilog
generate
if (ADDR_SIZE >= 8) begin end else $error("Must be 1024 KB minimum address size");
endgenerate
```

But this is not an immediate assertion - it will only be detected if a simulation is run, would like it to be detected at `Verilate` time or when it is compiled/converted to cpp files.

This would be a smaller chunk of the full assertion support - (also at runtime) but would be helpful to provide [SFINAE](https://en.wikipedia.org/wiki/Substitution_failure_is_not_an_error) safe code for systemverilog.

wsnyder, 2023-05-09:
Use

     if (ADDR_SIZE >= 8) ; else $fatal("Must be 1024 KB minimum address size");

This should be compatible everywhere, including synthesis (which asserts aren't), and is elaboration-time.

As to asserts themselves, we welcome a pull request to add the support!

wsnyder, 2025-09-05:
Missed this earlier, but this is invalid syntax, IEEE says immediate assertions must be procedural blocks.