verilator/verilator #4185: Config file for lint_off/on - selecting rules

svenka3, 2023-05-10:
I am trying to use Verilator with a subset of lint rules. Say I want only WIDTHEXPAND to be flagged. I create a .vlt file as below:

```config

`verilator_config
lint_off 
lint_on -rule WIDTHEXPAND 
```
But I don't get any warning on a code as below:
```verilog
module m;
  wire [3:0] bits_4;
  wire [7:0] bits_8;
  assign bits_8 = bits_4;
endmodule 

```
If I add a global lint_on as shown below, it works. 
```config

`verilator_config
lint_off 
lint_on -rule WIDTHEXPAND 
lint_on
```

Log:
```log
%Warning-WIDTHEXPAND: ../sv_tests/af_sv_mod.sv:6:17: Operator ASSIGNW expects 8 bits on the Assign RHS, but Assign RHS's VARREF 'bits_4' generates 4 bits.
```
So my question is - how do I turn-off all rules and selectively enable a few?

wsnyder, 2023-05-10:
lint_off is same as a waiver "lint_off *".

Waivers add a filter to all errors after the error is created.  So you can't turn back "on" a waiver once waived.

What you are trying to do makes sense; if you used the command line "-Wno-lint -WFOO" it would work.

I'm not sure what the right fix here is.

We could have lint_off/on without a file act like the command line, but that's inconsistent with how the same lint_off/on works when given a message.

We could keep matching the waivers on a given message, in order provided, and if it is later reenabled print it.  This perhaps is most intuitive, but requires reworking how the rules work as they are not retained/processed in order presently.

@wallento any opinions?

wallento, 2023-05-10:
I agree on your notion that this should be supported. I will have a look at how to implement it, probably not too complex, just that we must collect the order and match along the order.

svenka3, 2023-05-10:
Thanks. If you believe you can delegate it, I can attempt as well - though will need some directions (and at times it is easier/faster than explaining it to others, so I fully understand if you want to do it yourself @wallento ).

BTW based on @wsnyder suggestion (and few trials) above, I now have it working with the below command:

```sh
verilator --lint-only ../sv_tests/af_sv_mod.sv -Wno-lint -Wwarn-WIDTHEXPAND
```
Not pretty, not scalable, but works for now.

svenka3, 2023-05-25:
Gentle ping - in case you had a chance to look at this part of code. I see verilog.y having some relevant code as:
```yacc
  vltOffFront
                        { V3Config::addIgnore($1, false, "*", 0, 0); }
```
And calls to addWaiver in same place - is this the right place to start looking at please?

Thanks

svenka3, 2023-05-25:
Below is what I tried (without much success though).

I believe the flow working if I reverse the order - i.e. ON every rule, selectively turn-OFF. I believe this is because on seeing lint_off -rule, we call V3Config::addWaiver () - which inturn updates m_waivers.push_back(). 

Now I added a deleteWaiver method as below:

```cpp
    void deleteWaiver(V3ErrorCode code, const string& match) {
      Waivers::iterator wv_itr;  
        wv_itr = std::find(m_waivers.begin(), 
          m_waivers.end(), std::make_pair(code, match));
        if (wv_itr != m_waivers.end()) {
          m_waivers.erase (wv_itr);
          cout << " deleteWaiver \n";
        }
```
I call that deleteWaiver in verilog.y as 

```yacc
 vltOnFront yVLT_D_FILE yaSTRING yVLT_D_MATCH yaSTRING
                        { 
                              V3Config::deleteWaiver($1, *$3, *$5);
                        } 

```
I believe this is where I am stuck - when I have lint_on -rule ID, the above production does not kick-in.

Sample VLT file as below:
```verilog
`verilator_config
lint_off 
lint_on -rule WIDTHEXPAND 
// lint_off -rule ALWCOMBORDER 
// lint_on 
```

Am I approaching it the right way @wallento @wsnyder

wsnyder, 2023-05-26:
I think you need to be able to add 2 kinds of waivers to the lists, "on" and "off" waivers.  (Currently only have "off" ones).  The current code then scans the waivers and the first "hit" turns off the warnings and breaks out of the loop.  Instead, set a flag to "off" and if a later waiver turns it on, set the flag on, etc.  At the end, look at the on/off flag.

As to the grammar the line you selected matches "lint_off --file X --match Y" there's other lines above that which are used with just "lint_off --file" or other shorter versions with fewer flags.

svenka3, 2023-06-27:
Just a quick note - I don't believe I followed your latest comment. As I mentioned I see addWaiver() as most relevant method as start-point. But your latest comment seems to rethink the way we do waivers. I need more direction if I have to attempt, sorry if that's more than expected!

But I would definitely love to see this "feature" working soon :-) 

Thanks

closed 2025-12-31 by wsnyder