verilator/verilator #4190: Incorrect release of forced assignment

kbieganski, 2023-05-11:
Verilator version: befb415f278af51fc3ecc3dda1dfe820de023bd1
Test case:
```systemverilog
// verilator lint_off MULTIDRIVEN
module t (clk);
  input clk;

  logic [31:0] lhs1, lhs2, rhs;
  logic cond = 0;

  always_comb lhs1 = rhs;
  assign lhs2 = rhs;

  always @(posedge clk) rhs = '1;

  always @(negedge clk) begin
    if (cond) begin
      force lhs1 = 'hdeadbeef;
      force lhs2 = 'hfeedface;
    end
    else begin
      release lhs1;
      release lhs2;
    end
  end

  int cyc = 0;
  always @(posedge clk) begin
    cyc <= cyc + 1;
    if (cyc == 0) cond <= 1;
    if (cyc == 3) cond <= 0;
    if (cyc > 0) begin
      $display("cyc=%0d, cond=%b, lhs1=%x, lhs2=%x", cyc, cond, lhs1, lhs2);
      if (cond) begin
        if (lhs1 != 'hdeadbeef) $stop;
        if (lhs2 != 'hfeedface) $stop;
      end
      else begin
        if (lhs1 != '1) $stop;
        if (lhs2 != '1) $stop;
      end
      if (cyc > 8) begin
        $write("*-* All Finished *-*\n");
        $finish;
      end
    end
  end
endmodule
```
Output:
```
cyc=1, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=2, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=3, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=4, cond=0, lhs1=deadbeef, lhs2=deadbeef
%Error: test.v:36: Verilog $stop
```
`lhs2` is set to `lhs1`'s override for some reason.
With `-O0` it's a bit better:
```
cyc=1, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=2, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=3, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=4, cond=0, lhs1=deadbeef, lhs2=ffffffff
%Error: test.v:36: Verilog $stop
```
What's interesting, if we swap the `lhs1` and `lhs2` declarations, we get a different result.
```systemverilog
  logic [31:0] lhs2, lhs1, rhs;
```
Without `-O0` it actually works:
```
cyc=1, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=2, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=3, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=4, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=5, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=6, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=7, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=8, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=9, cond=0, lhs1=ffffffff, lhs2=ffffffff
*-* All Finished *-*
- test.v:41: Verilog $finish
```
With `-O0` it's the same as the original code:
```
cyc=1, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=2, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=3, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=4, cond=0, lhs1=deadbeef, lhs2=ffffffff
%Error: test.v:36: Verilog $stop
```
Not sure what's going on here, looks to me like two bugs on top of one another (invalid release + misoptimization).

kbieganski, 2023-05-25:
@wsnyder I believe this is still not fixed? The one I fixed is #4183, which is a different bug.

gezalore, 2024-03-03:
The output of event based simulators for this is actually:
```
cyc=1, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=2, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=3, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=4, cond=0, lhs1=deadbeef, lhs2=ffffffff
cyc=5, cond=0, lhs1=deadbeef, lhs2=ffffffff
cyc=6, cond=0, lhs1=deadbeef, lhs2=ffffffff
cyc=7, cond=0, lhs1=deadbeef, lhs2=ffffffff
cyc=8, cond=0, lhs1=deadbeef, lhs2=ffffffff
cyc=9, cond=0, lhs1=deadbeef, lhs2=ffffffff
```
You are assigning a constant to rhs, so there is no change event to trigger the `always_comb`, and `assign`:
```
  always @(posedge clk) rhs = '1;
```

However, IEEE 1800-2023, 10.6.2 The force and release procedural statements:
```
When released, then if the variable is not driven by a continuous assignment and does not currently have an active assign procedural continuous assignment, the variable shall not immediately change value and shall maintain its current value until the next procedural assignment to the variable is executed
```

That's why `lhs2` return tho the value of `rhs`, while `lhs1` doesn't.

`assign` is a continuous assignment, `always_comb` isn't.

Current output of verilator is

```
cyc=1, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=2, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=3, cond=1, lhs1=deadbeef, lhs2=feedface
cyc=4, cond=0, lhs1=deadbeef, lhs2=ffffffff
cyc=5, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=6, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=7, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=8, cond=0, lhs1=ffffffff, lhs2=ffffffff
cyc=9, cond=0, lhs1=ffffffff, lhs2=ffffffff
```

Which sees the return of `lhs1` to `rhs` as Verilator executes always_comb when `rhs` *might have changed*, not when it *did change*.

Not sure we can do much about this short term.