verilator/verilator #4201: verilated_trace_imp undefined behavior - store to misaligned address

jlwehle, 2023-05-13:
I'm using:

  Verilator 5.010
  FreeBSD 12.4
  clang 13.0

and ran verilator like so:

  verilator --binary --trace --timing -CFLAGS "-fsanitize='address,undefined'" -LDFLAGS "-fsanitize='address,undefined'" rx_cdr_bbpd_tb.v rx_cdr_bbpd.v

Executing the resulting binary results in:

  /usr/local/geda/share/verilator/include/verilated_trace_imp.h:875:5: runtime error: store to misaligned address 0x606000000634 for type 'QData' (aka 'unsigned long'), which requires 8 byte alignment
0x606000000634: note: pointer points here
  00 00 00 00 be be be be  be be be be be be be be  be be be be be be be be  be be be be be be be be
              ^ 

Looking at verilated_trace_imp.h:875:5 gives:

  ... fullQData(uint32_t* oldp, QData newval, int bits)
  ...
  *reinterpret_cast<QData*>(oldp) = newval;

a little bit further is:

  ... fullDouble(uint32_t* oldp, double newval)
  ...
  *reinterpret_cast<double*>(oldp) = newval;

There appears to be the assumption that a "uint32_t *" can be used to modify
types which require a stricter alignment (i.e. QData aka uint64_t, double).
Perhaps using memcpy would be more appropriate?

wsnyder, 2023-05-13:
Makes sense.  I don't know why there isn't a warning from gcc, clang nor any of the formal tools we use.

I was worried a memcpy might make less optimal code, but it seems to be identically good in a simple function.  (Which means your warning example must differ in some respect, if we expect memcpy to make it go away.)

Might you submit a pull to fix these up?

jlwehle, 2023-05-14:
I believe the compilers have gotten fairly smart about inlining memcpy calls in optimal fashion.

I don't have much experience with git / github, however I'm willing to give it a go with regards to a pull request.

wsnyder, 2023-06-12:
Thanks for the fix.

closed 2023-06-12 by wsnyder