verilator/verilator #4203: Active region did not converge.

jlwehle, 2023-05-14:
I'm using:

  Verilator 5.010
  FreeBSD 12.4
  clang 13.0

and ran verilator on a test case extracted from a larger project like so:

  verilator --binary --trace --timing t_rst_didnotconverge.v

Executing the resulting binary results in:

  %Error: t_rst_didnotconverge.v:1: Active region did not converge.
  Aborting...
  Abort

I also tried:

  Verilator 5.011 devel rev v5.010-47-g949be301d

The simulation (as well as the larger project) does succeed using Icarus Verilog.

I'm willing to do additional debugging if pointed in the right direction.

[t_rst_didnotconverge.zip](https://github.com/verilator/verilator/files/11471171/t_rst_didnotconverge.zip)

wsnyder, 2023-05-14:
This works for me on the same revision on Ubuntu,  please double check the example you provided doesn't work.  Assuming not, it's a freebsd issue you'll need to debug as I don't use freebsd. I'd suggest debugging by comparing the code flows in a debugger in each OS.

jlwehle, 2023-05-15:
I downloaded the zip file and confirmed it doesn't work on FreeBSD.
 
I don't use Ubuntu, however I'll see if I can setup a virtual machine.

jlwehle, 2023-05-30:
On FreeBSD we get to:

  V3SenExprBuilder.h:121   it = m_prev.emplace(*scopeExprp, prevp).first;

with the proper value in prevp.  However at:

  V3SenExprBuilder.h:129    AstVarScope* const prevp = it->second;

prevp contains the wrong value.

The earlier call to emplace causes m_prev to be searched using equal_to which said that:

  VARSCOPE 0x41cb6560 <e2138#> {d6ao} @dt=0x41ca2b20@(G/w1)  TOP->t__DOT__clk [T] [scopep=0x41cb61e0] -> VAR 0x415f15c0 <e1638#> {d6ao} @dt=0x41ca2b20@(G/w1)  t__DOT__clk [CLK] [VSTATIC]  VAR

and:

  VARSCOPE 0x41cb6480 <e2135#> {d5ao} @dt=0x41ca2b20@(G/w1)  TOP->t__DOT__rst [T] [scopep=0x41cb61e0] -> VAR 0x415f1480 <e1637#> {d5ao} @dt=0x41ca2b20@(G/w1)  t__DOT__rst [CLK] [VSTATIC]  VAR

are equal so an entry for the new prevp wasn't created, instead an IT which points to TOP->t__DOT__clk (the existing m_prev entry) was returned. Presumably the hash wasn't first looked at since m_prev only had a single existing entry.

Works with the changes in my pull request which implements class specific same methods for AstVarScope, AstScope, and AstScope.

wsnyder, 2023-06-03:
Thanks for debugging further and providing the pull.

closed 2023-06-03 by wsnyder