verilator/verilator #4204: Verilator fails on long symbol name when built with GCC 13.1.0 archlinux

Risto97, 2023-05-14:
I have a weird bug, could also be a bug in GCC but I wanted to report it anyways.
Lately I upgraded my system and I got the new GCC 13.1.0. I was recompiling verilator after the update and I noticed weird errors on some of the designs.

```
package gpio_regblock_pkg;

    typedef struct {
        logic value;
    } GPIO_ODR__odr0_hwclr_cb852fb2_hwset_c201fea2__out_t;

    typedef struct {
        GPIO_ODR__odr0_hwclr_cb852fb2_hwset_c201fea2__out_t odr0;
    } __GPIO_ODR_odr0_226a8fa3_odr1_38ac8d61_odr10_2c841a0c_odr11_4989e4b2_odr12_ba5781f9_odr13_d2a5bdea_odr14_2f00530e_odr15_d473;

    typedef struct {
        __GPIO_ODR_odr0_226a8fa3_odr1_38ac8d61_odr10_2c841a0c_odr11_4989e4b2_odr12_ba5781f9_odr13_d2a5bdea_odr14_2f00530e_odr15_d473 odr;
    } gpio_regblock__out_t;
endpackage
```
Running this
```
verilator_bin --compiler gcc  -cc --Wno-fatal gpio_regblock_pkg.sv
```
I get this error
```
%Error: gpio_regblock_pkg.sv:16:9: syntax error, unexpected IDENTIFIER, expecting TYPE-IDENTIFIER
   16 |         __GPIO_ODR_odr0_226a8fa3_odr1_38ac8d61_odr10_2c841a0c_odr11_4989e4b2_odr12_ba5781f9_odr13_d2a5bdea_odr14_2f00530e_odr15_d473 odr;
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: Exiting due to 1 error(s)
```
If I run the same thing with the same commit hash of verilator but compiled with GCC 12.1.0 it works without problems.

Now the problem is this string:
```
__GPIO_ODR_odr0_226a8fa3_odr1_38ac8d61_odr10_2c841a0c_odr11_4989e4b2_odr12_ba5781f9_odr13_d2a5bdea_odr14_2f00530e_odr15_d473
```
Originally it was longer but I cut it down to the 124 characters, and if I remove one character from the string it will compile.

What OS and distribution are you using?
Archlinux GCC 13.1.0

May we assist you in trying to fix this in Verilator yourself?
Sure I can try. But this might not be a Verilator problem at all, it could just be GCC.
But as I didn't notice any other problem with my system apart from this, I report it here.

wsnyder, 2023-05-15:
Verilator does shorten names internally, but given it works with a different gcc, either gcc or glibc (which comes with gcc) seems suspect.  You might try Verilating with "--debug" to see if you get a better error, as the debug build should have compiled with _GLIBCXX_DEBUG which adds glibc assertions.

Risto97, 2023-05-15:
I tested it on Ubuntu 20.04, compiling the GCC13.1.0, and it still fails.
With system GCC  (9.4.0) it works...

I can see that the -D_GLIBCXX_DEBUG was present during the compilation.
I didn't set the LD_LIBRARY_PATH to the new GLIBC so I guess it took the Ubuntu system one.
Also I don't see it looking for libstdc++ with ldd, so I guess it's linked statically, glibc is showing as system one through ldd.

Meaning that GLIBC is not the problem, but its the newer GCC.

Now the weird thing is that running --debug I don't get the error, instead, it does the same thing as with the working GCC version.
Diffing the directory produced by --debug with the working one, I don't see any notable differences, it's a bit hard to diff it because of the pointer information in the output, which will always differ.

[working.tar.gz](https://github.com/verilator/verilator/files/11474990/working.tar.gz)
[broken.tar.gz](https://github.com/verilator/verilator/files/11474989/broken.tar.gz)

I will try to dive a bit deeper when I have time.
If you have any hints about where should I look, let me know.

closed 2023-05-27 by wsnyder

Risto97, 2023-08-03:
@wsnyder 
Well this is still an issue.

Something happens during parsing and long symbols gets shorted to some weird hash value.
It even thinks there is a another module on the example below.
```verilog
typedef struct {
    logic odr0;
} aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa;

module test;
    aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa odr;
endmodule

```

Running:
`verilator_bin --cc   -Wno-fatal test.sv  > log.txt`

```
%Error: test.sv:6:5: Cannot find file containing module: 'aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa__Vhsh1WtDVAABQQUf1eFRJ29kA6Os61ycPXitVeYdQDMw'
    6 |     aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa odr;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        ... Unsupported: Name is longer than 127 characters; automatic file lookup not supported.
        ... Suggest putting filename with this module/package onto command line instead.
%Error: Exiting due to 1 error(s)
```

Running the same thing with `verilator_bin_dbg` works correctly.

Here is a docker image that proves there is a problem.
[Dockerfile.txt](https://github.com/verilator/verilator/files/12254622/Dockerfile.txt)

I was trying to trace the bug with ` --debugi 9` enabled, and I saved the log both with verilator compiled with GCC12 and the one with GCC13.

There is a weird problem, it seems like that the long string "aaaaa...." is shortened differently twice.
First time I see:
`aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa__VhshCaDfuPimwBvFlqXoTcLXAcShuUvu7WGnlqeFNdgE`
Then I see:
`aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa__VhshT4X1zowSaCBjNAwL3XhJBTzoaaQvOkAASTMPIo2P`
While in the working log there is only one version.

What happens is that this function find call does not find whatever it is looking for in the broken version:
https://github.com/verilator/verilator/blob/9caa79a7eac1fd9164c5ff69602dbaab1f4e0663/src/V3ParseImp.cpp#L576

Then if you search for `&&&&&&&& FOUNDP: 0x55cc029a4b80` in compare_to_this.txt (working log)
There is a line:
```
tokenToBison  TOKEN {d6-6:5-137[ct3+9]}=262 "TYPE-IDENTIFIER" strp='aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa__VhshlJwPDDFrSAXxxsqPjlGtBmJrmUYdyNIRkmetp23c'
```
While in the non working log I have this:
```
tokenToBison  TOKEN {d6-6:5-137[ct3+9]}=103 "IDENTIFIER" strp='aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa__VhshT4X1zowSaCBjNAwL3XhJBTzoaaQvOkAASTMPIo2P'
```

[broken_log.txt](https://github.com/verilator/verilator/files/12255371/broken_log.txt)
[compare_to_this.txt](https://github.com/verilator/verilator/files/12255372/compare_to_this.txt)

I feel like I am getting close to the issue, but cannot really locate it.
I suspect now that the problem is in whatever is changing these long symbol names to these shorter strings.
I will have a look at it again, if you have a hint where should I look that would be really useful.

wsnyder, 2023-08-08:
I'd suggest add/enable debug prints in the hasher in V3String that print the pre and post-hash strings. Then run your good and bad examples and see what differs.