verilator/verilator #4207: Is there any way of building a fully static version of Verilator?

tudortimi, 2023-05-17:
A fully statically linked binary would allow it to be used without having to install any other dependencies. This would be advantageous for stuff like CI jobs, where it would be enough to just get the Verilator install tree, without any extra dependencies. GitHub has very nice caching capabilities for this.

I see there is a `--disable-partial-static` option for `./configure`, but I would need something like `--enable-fully-static`.

tudortimi, 2023-05-17:
Might this also be possible using `LDFLAG`?

furkanusta, 2023-05-17:
With CMake this works
```
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_EXE_LINKER_FLAGS=-static   -DCMAKE_FIND_LIBRARY_SUFFIXES=.a ..
```

This also seems to work but I am not sure if cmake is doing anything extra and this is missing an edge case
```
./configure LDFLAGS=-static
make
```

wsnyder, 2023-05-24:
Note also you can use VERILATOR_ROOT at runtime, as long as you configure without a VERILATOR_ROOT, and do NOT "make install".  see https://veripool.org/guide/latest/install.html#eventual-installation-options

closed 2023-05-24 by wsnyder