verilator/verilator #4209: Misordered assignments of registers when `-fno-split` is given

YikeZhou, 2023-05-18:
### Example code

Verilog file `top.v`:
```verilog
module top (O, clk, wire1, wire0);
  output [5:0] O;
  input clk;
  input wire [2:0] wire0;
  input wire [2:0] wire1;
  wire [2:0] wire2;
  wire [2:0] wire3;
  reg [2:0] reg0 /*verilator public_flat*/ = (2'h0);
  reg [2:0] reg1 /*verilator public_flat*/ = (2'h0);
  reg [2:0] reg2 /*verilator public_flat*/ = (2'h0);
  reg [2:0] reg3 /*verilator public_flat*/ = (2'h0);
  reg [2:0] reg4 /*verilator public_flat*/ = (2'h0);

  always @(posedge clk) begin
    if ((-wire1))
      reg0 <= wire1;
    else
      reg1 = wire0;
  end
  always @(posedge clk) begin
    reg2 <= {reg0};
  end
  assign wire3 = (wire0 == reg0);
  assign wire2 = {reg2, wire3, reg0, reg1};

  always @(posedge clk) begin
    if (wire0[1])
      reg4 <= (~&wire2);
    if ((wire2 * wire2))
      reg3 <= $unsigned(wire2);
    else
      reg3 <= reg4;
  end
  assign O = {reg3, reg4, wire2};
endmodule
```

C++ file `main.cpp`:
```cpp
#include "verilated.h"
#include "Vtop.h"

#include <bitset>
#include <iostream>

//======================

int main(int argc, char** argv, char**) {
    // Setup context, defaults, and parse command line
    Verilated::debug(0);
    const std::unique_ptr<VerilatedContext> contextp{new VerilatedContext};
    contextp->commandArgs(argc, argv);

    // Construct the Verilated model, from Vtop.h generated from Verilating
    const std::unique_ptr<Vtop> topp{new Vtop{contextp.get()}};

    // Simulate
    topp->clk = 0;
    topp->eval();
    topp->wire0 = 7;
    topp->wire1 = 0;
    topp->clk = 1;
    topp->eval();
    std::cout << "O = " << std::bitset<8>(topp->O) << std::endl;

    // Final model cleanup
    topp->final();
    return 0;
}
```

### Actual Behavior

**Without** the `-fno-split` option, `O = 00000111`.

```sh
$ verilator -cc -exe -Wno-fatal -Wno-lint -Mdir default -build top.v main.cpp && ./default/Vtop          
make: Entering directory '/home/zyk/C/verilator/default'
... # Omitted
make: Leaving directory '/home/zyk/C/verilator/default'
O = 00000111
```

**With** the `-fno-split` option, `O = 00001111`.

```sh
$ verilator -cc -exe -Wno-fatal -Wno-lint -fno-split -Mdir no_split -build top.v main.cpp && ./no_split/Vtop
make: Entering directory '/home/zyk/C/verilator/no_split'
... # Omitted
make: Leaving directory '/home/zyk/C/verilator/no_split'
O = 00001111
```

### Verilator version

The latest git master (`729f8b9334ce01d289e31e59a126f1611b20736b`).

### OS version

WSL2 on Windows (Debian 10)

YikeZhou, 2023-05-18:
Discrepancies were found in `Vtop___024root___nba_sequent__TOP__0` in the following two files.
- `default/Vtop___024root__DepSet_heccd7ead__0.cpp`: `top__DOT__reg1` was modified before being referenced.
- `no_split/Vtop___024root__DepSet_heccd7ead__0.cpp`: `top__DOT__reg1` was referenced and then modified.

```cpp
// The "default" version
VL_INLINE_OPT void Vtop___024root___nba_sequent__TOP__0(Vtop___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___nba_sequent__TOP__0\n"); );
    // Body
    if ((1U & (~ (IData)((0U != (7U & (- (IData)(vlSelf->wire1)))))))) {
        vlSelf->top__DOT__reg1 = vlSelf->wire0;  // ■■■■■ reg1 = wire0 ■■■■■
    }
    vlSelf->top__DOT__reg2 = vlSelf->top__DOT__reg0;
    vlSelf->top__DOT__reg3 = ((0U != (7U & ((IData)(vlSelf->top__DOT__reg1) 
                                            * (IData)(vlSelf->top__DOT__reg1))))
                               ? (IData)(vlSelf->top__DOT__reg1)
                               : (IData)(vlSelf->top__DOT__reg4));
    if ((0U != (7U & (- (IData)(vlSelf->wire1))))) {
        vlSelf->top__DOT__reg0 = vlSelf->wire1;
    }
    if ((2U & (IData)(vlSelf->wire0))) {
        vlSelf->top__DOT__reg4 = (1U & (~ (IData)((7U 
                                                   == (IData)(vlSelf->top__DOT__reg1)))));
    }
    vlSelf->O = (((IData)(vlSelf->top__DOT__reg4) << 3U) 
                 | (IData)(vlSelf->top__DOT__reg1));
}
```

```cpp
// The "-fno-split" version
VL_INLINE_OPT void Vtop___024root___nba_sequent__TOP__0(Vtop___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___nba_sequent__TOP__0\n"); );
    // Body
    vlSelf->top__DOT__reg3 = ((0U != (7U & ((IData)(vlSelf->top__DOT__reg1) 
                                            * (IData)(vlSelf->top__DOT__reg1))))
                               ? (IData)(vlSelf->top__DOT__reg1)
                               : (IData)(vlSelf->top__DOT__reg4));
    if ((2U & (IData)(vlSelf->wire0))) {
        vlSelf->top__DOT__reg4 = (1U & (~ (IData)((7U 
                                                   == (IData)(vlSelf->top__DOT__reg1)))));
    }
    vlSelf->top__DOT__reg2 = vlSelf->top__DOT__reg0;
    if ((0U != (7U & (- (IData)(vlSelf->wire1))))) {
        vlSelf->top__DOT__reg0 = vlSelf->wire1;
    } else {
        vlSelf->top__DOT__reg1 = vlSelf->wire0;  // ■■■■■ reg1 = wire0 ■■■■■
    }
    vlSelf->O = (((IData)(vlSelf->top__DOT__reg4) << 3U) 
                 | (IData)(vlSelf->top__DOT__reg1));
}
```

wsnyder, 2023-05-18:
Two problems with your test.

First, out[3] which is the bit that changes comes from a posedge clk that uses wire2, which uses reg0, which comes from a flop. So until you have two clock edges the results are unpredictable (would be X if using a 4-state simulator).  So this test is dependent on initial state which is confusing.

Second, you have

         reg1 = wire0;  // <= 0

Not <=. So the simulator can choose either ordering in setting reg1.  It happens that -fno-split or not makes two different choices, but due to the blocking assignment both are legal by IEEE.

Note if you use -Wall Verilator tells you about this.

```
%Warning-BLKSEQ: t/t_0.v:33:14: Blocking assignment '=' in sequential logic process
                              : ... Suggest using delayed assignment '<='
   33 |         reg1 = wire0;
      |              ^
```

closed 2023-05-18 by wsnyder

YikeZhou, 2023-05-19:
Thank you very much for the detailed explanation! 

Sorry to bother you. I shall check all the warnings before sending an issue next time. 😔