verilator/verilator #4211: VERILATOR_ROOT is inconsistent

Timmmm, 2023-05-19:
Using v4.210 but I don't believe anything has changed relative to `master` from today.

If you are using an install of Verilator that someone has moved from its `--prefix` (or maybe you want to use a portable one) then you might think you can just set `VERILATOR_ROOT=/path/to/verilator_install` and be done, but unfortunately that doesn't work.

This is because when running `verilator`, it will look for `verilator_bin` in `$VERILATOR_ROOT/bin/verilator_bin` (among other places). This implies you should set `VERILATOR_ROOT=/path/to/verilator_install`. That will work for the `verilator` run.

However when you then do `make -f Vtop.mk` it will try to load `$VERILATOR_ROOT/include/verilated.mk`. Unfortunately this is *not* installed at `/path/to/verilator_install/include/verilated.mk`, but `/path/to/verilator_install/share/verilator/include/verilated.mk`.

Ok so you can change `VERILATOR_ROOT` to point to `/path/to/verilator_install/share/verilator` but then `verilator` won't run!

Fundamentally you need a different value of `VERILATOR_ROOT` for `verilator` and `make`, which is very annoying and impossible in some cases.

I think the solution is in `V3EmitMk.cpp` change

```
        of.puts("include $(VERILATOR_ROOT)/include/verilated.mk\n");
```

to

```
        of.puts("include $(VERILATOR_ROOT)/share/verilator/include/verilated.mk\n");
```

Or maybe to ditch the logic that uses `VERILATOR_ROOT` in the `verilator` Perl script (I don't know why you'd ever want anything except "use the verilator_bin in the same directory as this" which it can already do).

wsnyder, 2023-05-19:
Summarizing what is described here https://veripool.org/guide/latest/install.html#eventual-installation-options you can either
1. configure with prefix, make install, then run from the installed dir (without VERILATOR_ROOT)
2. configure, don't install, run with VERILATOR_ROOT pointing at the source directory

Mixing VERILATOR_ROOT with make install is not supported, because the various distros have different ways they require installed files be made, and because option 2 seems sufficient for anyone that wants it "relocatable"

This might not be clear in the documentation, if not we welcome a pull request to the documentation to make this clearer.

closed 2023-05-19 by wsnyder

Timmmm, 2023-05-20:
Ah I see, that was indeed not at all clear! I'll try to make a PR if I get time.