verilator/verilator #4213: share/verilator/include/verilated.mk contains non-existent AR=/usr/local/bin/ar on FreeBSD

yurivict, 2023-05-21:
On FreeBSD it installs the file ```share/verilator/include/verilated.mk``` with this line:
>  AR = /usr/local/bin/ar

The build should find the ar executable and use it, for example:
```
$ which ar
/usr/bin/ar
```

Version: 5.010
FreeBSD 13.2 STABLE

christopher-bowman, 2023-05-21:
yes making the change in share/verilator/include/verilated.mk seems to work for me (I just didn't know where to look, thank you)

igalic, 2023-05-21:
it might be good to change this to

```majefike
AR ?= /usr/bin/ar
```

i reckon many package build systems might set that variable

yurivict, 2023-05-21:
I think that this line should be just:
> AR = ar

to allow the PATH variable to determine the path.

wsnyder, 2023-05-22:
Note it comes from whatever ./configure determines.  You can override it with "./configure AR=AR".

yurivict, 2023-05-22:
> Note it comes from whatever ./configure determines. You can override it with "./configure AR=ar".

Then this is a bug in configure because ```/usr/local/bin/ar``` doesn't exist on this platform.

wsnyder, 2023-05-22:
What does ./configure print?

     checking for ar... ar

You can add echos or whatever to debug the shell code in "configure" that computes this.  Most likely you do have a $AR variable, or ar is in your $PATH. This code shouldn't be able to pull "/usr/local/bin" from nowhere.

yurivict, 2023-05-22:
I found what the problem was.

For whatever reason verilator's port build requires ar from binutils. But binutils wasn't present at the runtime, so /usr/local/bin/ar provided by binutils isn't present.

Some time back binutils was added as dependency to solve some problem, I don't remember what problem that was.

I will correct the port.

Thank you for your help!

closed 2023-05-22 by yurivict