verilator/verilator #4215: Linting produced error when `-I` is used

hpretl, 2023-05-22:
Using `Verilator` in the `OpenLane` flow produces a linting fail, which is documented here: https://github.com/The-OpenROAD-Project/OpenLane/issues/1811

It seems like `verilated_std.sv` is auto-included multiple times, triggering an error the 2nd time.

Verilator version: 947402bc57625106e1387255772fdb5d850a6c65

wsnyder, 2023-05-22:
Can you please provide a simple self-contained (e.g. needing no install) example showing the issue?

hpretl, 2023-05-22:
Maybe not the best example, but it triggers the error: `verilator --lint-only -Wall spm.v -I spm.v`

[spm.v.zip](https://github.com/verilator/verilator/files/11534730/spm.v.zip)

wsnyder, 2023-05-23:
This is happening because you gave "-I " which means use an include directory of "nowhere", which is somewhat nonsensical.

Regardless the error is certainly confusing, it's because the -I confused where it thought verilated_std.vh lived. Fixed.

closed 2023-05-23 by wsnyder