verilator/verilator #4221: Wrong latch inference when using function in always_comb block

adrienlemasle, 2023-05-24:
With latest version of Verilator, the following code:

```
module verilator_latch
  (
   input logic [2:0] state,
   output logic [31:0] b
   );

   function logic [31:0 ] toto () ;
      logic [31:0] res;
      res = 10;

      return res;
   endfunction

   always_comb begin
      b = 0;
      if(state != 2) begin
         b = toto();
      end
   end

endmodule;
```

Gives the following warning:
```
verilator --no-timing --lint-only -Wall verilator_latch.sv
%Warning-LATCH: verilator_latch.sv:14:4: Latch inferred for signal 'verilator_latch.toto__Vstatic__res' (not all control paths of combinational always assign a value)
                                       : ... Suggest use of always_latch for intentional latches
   14 |    always_comb begin
      |    ^~~~~~~~~~~
                ... For warning description see https://verilator.org/warn/LATCH?v=5.010
                ... Use "/* verilator lint_off LATCH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

There is no actual latch in this code so this seems to be a bug to me.

wsnyder, 2023-05-24:
A function can retain state, and that state is not retained if the "if (state" is false.

Use "function automatic" instead of "function".

In theory Verilator could analyze the function more completely to determine that the return value is never consumed in any way that requires the state be retained, then suppress the LATCH warning, but this is difficult.

margej, 2023-06-09:
I've raised a PR to address this issue, please take a look
https://github.com/verilator/verilator/pull/4284

wsnyder, 2025-09-09:
#4284 pull fixed this (thanks).  Confirmed still passes on current master.

closed 2025-09-09 by wsnyder