verilator/verilator #4225: export DPI calls with timing controls are broken

udif, 2023-05-24:
> Thanks for taking the time to report this.

> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

dpi_test.v:
```
module dpi_test ();

reg rtl_clk;
initial begin
    rtl_clk = 1'b0;
    forever
        #2 rtl_clk = ~rtl_clk;
end

    export "DPI-C" v_export = task dpi_export;
    task dpi_export(input int unsigned i);
        //@(posedge rtl_clk); // comment this out for the task to be executed
        $display("%t: v_export: i=%3d", $time, i);
    endtask

    import "DPI-C" context task dpi_import(input int unsigned n);

    integer n;
initial begin
        $display("Dumping waveforms");
        $dumpfile("waves.fst");
        repeat (10) @(posedge rtl_clk);
        for (n = 3; n < 6; n = n + 1) begin
            $display("%t: calling dpi_import: n =%3d", $time, n);
            dpi_import(n);
        end
        $finish;
end
endmodule
```
test.c
```
#include <assert.h>
#include <stdio.h>
#include <svdpi.h>

#ifdef __cplusplus
extern "C" {
#endif

extern void v_export (unsigned int i);

int dpi_import(unsigned int len) {
    printf("dpi_import: len=%d\n", len);
    for (int i = 0; i < len; i++) {
        printf("callling v_export: i=%d\n", i);
        v_export(i);
    }
    return 0;
}

#ifdef __cplusplus
}
#endif
```
Compiled using:
```
verilator --binary --trace-fst dpi_test.v test.c
obj_dir/Vdpi_test
```

Yields:
```
Dumping waveforms
                  38: calling dpi_import: n =  3
dpi_import: len=3
callling v_export: i=0
                  38: v_export: i=  0
callling v_export: i=1
                  38: v_export: i=  1
callling v_export: i=2
                  38: v_export: i=  2
                  38: calling dpi_import: n =  4
dpi_import: len=4
callling v_export: i=0
                  38: v_export: i=  0
callling v_export: i=1
                  38: v_export: i=  1
callling v_export: i=2
                  38: v_export: i=  2
callling v_export: i=3
                  38: v_export: i=  3
                  38: calling dpi_import: n =  5
dpi_import: len=5
callling v_export: i=0
                  38: v_export: i=  0
callling v_export: i=1
                  38: v_export: i=  1
callling v_export: i=2
                  38: v_export: i=  2
callling v_export: i=3
                  38: v_export: i=  3
callling v_export: i=4
                  38: v_export: i=  4
- dpi_test.v:27: Verilog $finish
```

Which is fine.

Uncommenting out dpi_test.v:12 (event delay) and rerunning, yields:
```
Dumping waveforms
                  38: calling dpi_import: n =  3
dpi_import: len=3
callling v_export: i=0
callling v_export: i=1
callling v_export: i=2
                  38: calling dpi_import: n =  4
dpi_import: len=4
callling v_export: i=0
callling v_export: i=1
callling v_export: i=2
callling v_export: i=3
                  38: calling dpi_import: n =  5
dpi_import: len=5
callling v_export: i=0
callling v_export: i=1
callling v_export: i=2
callling v_export: i=3
callling v_export: i=4
- dpi_test.v:27: Verilog $finish
```

Running the same code with a Big-3 simulator yields the correct result:
```
Dumping waveforms
                  38: calling dpi_import: n =  3
dpi_import: len=3
callling v_export: i=0
                  42: v_export: i=  0
callling v_export: i=1
                  46: v_export: i=  1
callling v_export: i=2
                  50: v_export: i=  2
                  50: calling dpi_import: n =  4
dpi_import: len=4
callling v_export: i=0
                  54: v_export: i=  0
callling v_export: i=1
                  58: v_export: i=  1
callling v_export: i=2
                  62: v_export: i=  2
callling v_export: i=3
                  66: v_export: i=  3
                  66: calling dpi_import: n =  5
dpi_import: len=5
callling v_export: i=0
                  70: v_export: i=  0
callling v_export: i=1
                  74: v_export: i=  1
callling v_export: i=2
                  78: v_export: i=  2
callling v_export: i=3
                  82: v_export: i=  3
callling v_export: i=4
                  86: v_export: i=  4
Simulation complete via $finish(1) at time 86 NS + 0
./dpi_test.v:27         $finish;
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.011 devel rev v5.010-1-g3e986517c
This is a very recent version although not the current head.

> What OS and distribution are you using?

Ubuntu 20.04

> May we assist you in trying to fix this in Verilator yourself?

Given a hint I could try, although this doesn't sound simple.

wsnyder, 2023-05-24:
I think it's unlikely we will support this any time soon, but perhaps @kbieganski has an idea how to do it more easily.

At minimum we should add an unsupported warning on any timing elements in a DPI export.

udif, 2023-05-24:
Perhaps this should be documented in the DPI section, it would save time for other people not to try this.

I had another issue in the main test (this is just a synthetic example inspired by a large SoC simulation), where Verilator failed to build a default context for DPI imports, and crashed until I built the context manually, but I cannot reproduce it with this minimal example.

wsnyder, 2023-05-24:
Yes, though I think it should warn you directly when it occurs, as I suspect most do not read the documentation ;)

kbieganski, 2023-05-25:
It's not a trivial problem. Blocking in Verilator is done using coroutines, which don't actually block anything, they just pass control to a different place in the code. When you call an exported function, coroutines deeper in the stack actually do get suspended. However, the exported function just finishes and returns control to your code, which proceeds to call it again in a loop.

In order for an exported function to block, it has to block the calling thread in some way. This is certainly possible, but you'd need a way to subscribe to the awaited event in the DPI wrapper generated by Verilator and then unblock when the event is triggered. I think it's doable by passing the necessary machinery via the coroutine return type (`VlCoroutine`).

That's not so bad, but it gets difficult if you take into account calling a DPI export from a DPI import. Now you have a simulation thread calling your code, which calls a blocking function, which results in the simulation thread being blocked. This is likely (or always?) the thread that's supposed to resume your coroutine, and thus unblock the thread (so unblock itself). Thus we get a deadlock. At this point I don't see a good way to tackle that issue.

I'll add a warning about this being unsupported.

kbieganski, 2023-06-07:
A warning was added in https://github.com/verilator/verilator/pull/4238. Rescoping this issue to adding support for DPI exports with timing.

closed 2026-11-12 by wsnyder