verilator/verilator #4226: [Linting] Wrong multidriven warning on unpacked structures

Risto97, 2023-05-24:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```verilog
module test;
    typedef struct {
        logic val1;
        logic val2;
    } str_t;
    str_t str;

    always_comb begin
        str.val1 = 1'b0;
    end

    always_comb begin
        str.val2 = 1'b0;
    end
endmodule
```
Results in:

```
%Warning-MULTIDRIVEN: test.sv:13:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'str'
                                  : ... In instance test
                      test.sv:13:9:
   13 |         str.val2 = 1'b0;
      |         ^~~
                      test.sv:9:9: ... Location of other write
    9 |         str.val1 = 1'b0;
      |         ^~~
                      ... For warning description see https://verilator.org/warn/MULTIDRIVEN?v=5.011
                      ... Use "/* verilator lint_off MULTIDRIVEN */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

It does not happen if the struct is packed.

What 'verilator --version' are you using?  Did you try it with the git master version?
Verilator 5.011 devel rev v5.010-30-g0606a29f1

May we assist you in trying to fix this in Verilator yourself?
Sure

Risto97, 2023-05-24:
BTW is this the preferred way of attaching the test case?

[t_multidriven_unpacked_struct.tar.gz](https://github.com/verilator/verilator/files/11559114/t_multidriven_unpacked_struct.tar.gz)

wsnyder, 2023-05-24:
The multidriven warnings do consider each bit separately (which is why packed works), but doesn't currently analyze unpacked structure members (or many other complicated things like multidimensional unpacked array elements).  We'd welcome a pull to enhance this area, I can discuss some ideas if you want to work on it.

Risto97, 2023-05-25:
This might be the same or a different issue, for now I leave it here, but I can create a new issue for this.

```systemverilog
module test (
        input wire clk,
        output logic  q
    );

    typedef struct {
        logic overflow2;
        logic next;
    } comb_t;
    comb_t comb;

    always_comb begin
        /* comb.next = 0; */
        if(comb.overflow2) begin 
            comb.next = 1;
        end else begin
            comb.next = 0;
        end
    end

    always_ff @(posedge clk) begin
        q <= comb.next;
    end
endmodule
```
It gives out warnings, that should not exist due to previous issue .

```
%Warning-ALWCOMBORDER: test.sv:15:13: Always_comb variable driven after use: 'comb'
                                    : ... In instance test
   15 |             comb.next = 1;
      |             ^~~~
                       ... For warning description see https://verilator.org/warn/ALWCOMBORDER?v=5.011
                       ... Use "/* verilator lint_off ALWCOMBORDER */" and lint_on around source to disable this message.
%Warning-UNOPTFLAT: test.sv:10:12: Signal unoptimizable: Circular combinational logic: 'test.comb'
   10 |     comb_t comb;
      |            ^~~~
                    test.sv:10:12:      Example path: test.comb
                    test.sv:12:5:      Example path: ALWAYS
                    test.sv:10:12:      Example path: test.comb
```

But additionally it fails to compile and I get this error:
`verilator_bin  --build  --cc  --Wno-fatal  test.sv`
```
Vtest___024root__DepSet_h9b2d61ff__0.cpp: In function ‘void Vtest___024root___eval_triggers__act(Vtest___024root*)’:
Vtest___024root__DepSet_h9b2d61ff__0.cpp:21:38: error: no match for ‘operator!=’ (operand types are ‘Vtest_test__DOT__comb_t__struct3’ and ‘Vtest_test__DOT__comb_t__struct3’)
   20 |     vlSelf->__VactTriggered.set(0U, (vlSelf->test__DOT__comb
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~
      |                                              |
      |                                              Vtest_test__DOT__comb_t__struct3
   21 |                                      != vlSelf->__Vtrigprevexpr___TOP__test__DOT__comb__0));
      |                                      ^~ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 Vtest_test__DOT__comb_t__struct3
In file included from Vtest__ALL.cpp:7:
Vtest___024root__DepSet_h9b2d61ff__0__Slow.cpp: In function ‘void Vtest___024root___eval_triggers__stl(Vtest___024root*)’:
Vtest___024root__DepSet_h9b2d61ff__0__Slow.cpp:22:38: error: no match for ‘operator!=’ (operand types are ‘Vtest_test__DOT__comb_t__struct3’ and ‘Vtest_test__DOT__comb_t__struct3’)
   21 |     vlSelf->__VstlTriggered.set(1U, (vlSelf->test__DOT__comb
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~
      |                                              |
      |                                              Vtest_test__DOT__comb_t__struct3
   22 |                                      != vlSelf->__Vtrigprevexpr___TOP__test__DOT__comb__0));
      |                                      ^~ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 |
      |                                                 Vtest_test__DOT__comb_t__struct3
```
What is interesting that it does not treat default value for `comb.next` the same way if it is in else begin block or at the beginning of always block, if you uncomment the commented line, it will work.
This is what makes me think that this test case is maybe exposing a separate bug.

@wsnyder I can give it a try when I have time. If you have time you can write me some hints on how to solve it.

wsnyder, 2023-05-26:
The unopt flat is expected as it schedules entire variables.  The compile error is something else, could you please make a new ticket for that (including the verilator lint_off on the warnings)?  Thanks

Risto97, 2023-05-26:
@wsnyder done #4234