verilator/verilator #4227: Getting VCD syntax error of generated vcd in Cadence tools

AnikBalo, 2023-05-25:
Verilator version: 4.110
OS: Ubuntu-20.04
I performed RTL simulation using verilator and generated a VCD of that RTL [ https://github.com/openhwgroup/cva6 ]  simulations without any error. While using that VCD in cadence tools, the tool saying there is syntax error at line 114 in VCD file.

The VCD file snapshot: 
[image]

Can anyone help what's wrong here? 
Thanks!

wsnyder, 2023-05-26:
"struct" isn't legal in VCD's it needs to be "module".  Perhaps you could create a pull request to add this to e.g. t_trace_complex.v and fix verilated_vcd.cpp to only print "module"?

atorregrosa-smd, 2023-05-26:
Actually, this was introduced in #2776 & #2805, as [GTKWave parses it that way](https://github.com/gtkwave/gtkwave/blob/master/gtkwave3-gtk3/src/helpers/vcd2fst.c#L887) to show a struct icon in the hierarchy tree. 

The LRM does however say that the only supported scope types are `module`, `task`, `function`, `begin` and `fork`, so this relies on non-standard behavior. It's probably better then to leave the gtkwave-specific types just in `.fst` waves.

wsnyder, 2023-05-27:
@atorregrosa-smd, agreed, would you be willing to make a pull request to fix this please?

atorregrosa-smd, 2023-05-30:
Yes, no problem.

PR #4282 (2023-06-08): Fix VCD scope types
This limits the custom struct/union/interface scope types to be issued only in FST traces, since they are non-standard for VCD.

Should fix #4227