verilator/verilator #4232: Disable automatic file searching

Timmmm, 2023-05-25:
Currently if one of your SV source files contains a module:

```
module top;
   foo u_foo;
endmodule;
```

Verilator will automatically search for `foo`, `foo.v` and `foo.sv` in all the include paths. It outputs something like this:

```
%Error: our.v:3:6: Cannot find file containing module: 'foo'
    3 |      foo u_foo;
      |      ^~~
%Error: our.v:3:6: This may be because there's no search path specified with -I<dir>.
    3 |      foo u_foo;
      |      ^~~
        ... Looked in:
             foo
             foo.v
             foo.sv
             obj_dir/foo
             obj_dir/foo.v
             obj_dir/foo.sv
```

Is there any way to turn this behaviour off? I still need to have search paths for some `include` directives so "don't use `-I` isn't a possible answer unfortunately.

Btw there doesn't seem to be any documentation about this feature, or how SV compilation units work in general (are all the files a single compilation unit?)

wsnyder, 2023-05-26:
I don't understand what you mean by "turn off".  How do you want "module foo" to be resolved otherwise?

Timmmm, 2023-05-27:
I want it to already know about that module because it was part of a file I listed earlier in the explicit list of files to compile.

I'm pretty sure that's how Questa, VCS and Xcelium work.

If you write `foo f;` in C++ and it doesn't know the type `foo`, it doesn't automatically try to open `foo.h`.

Does that make sense? Hopefully I haven't misunderstood anything!

wsnyder, 2023-05-27:
That message means the module was *not* found in any files you previously listed; if you believe you did please provide a complete self-contained example showing the problem.

I believe the module resolution is similar to all other simulators.

Timmmm, 2023-05-27:
The difference is that it *would* have loaded `foo.sv` if it existed *even if I didn't explicitly list it*. That's unexpected. Here's an example.

In `top.sv`:

```
module top(input logic clk, input logic in, output var logic out);
  foo u_foo(in, out);
endmodule
```

In `foo.sv`:

```
module foo(input logic a, output var logic b);
    always_comb b = ~a;
endmodule
```

Now if we compile it with both files in Questa it works fine as expected:

```
❯ vlog foo.sv top.sv && vopt top -o top_opt
QuestaSim-64 vlog 2022.1_1 Compiler 2022.02 Feb 19 2022
Start time: 22:19:06 on May 27,2023
vlog foo.sv top.sv 
-- Compiling module foo
-- Compiling module top

Top level modules:
        top
End time: 22:19:06 on May 27,2023, Elapsed time: 0:00:00
Errors: 0, Warnings: 0
QuestaSim-64 vopt 2022.1_1 Compiler 2022.02 Feb 19 2022
Start time: 22:19:06 on May 27,2023
vopt top -o top_opt 

Top level modules:
        top

Analyzing design...
-- Loading module top
-- Loading module foo
Incremental compilation check found no design-units have changed.
Optimized design name is top_opt
End time: 22:19:06 on May 27,2023, Elapsed time: 0:00:00
Errors: 0, Warnings: 0
```

However if we omit `foo.sv` it *won't automatically look for it*:

```
❯ vlog top.sv && vopt top -o top_opt
QuestaSim-64 vlog 2022.1_1 Compiler 2022.02 Feb 19 2022
Start time: 22:19:47 on May 27,2023
vlog top.sv 
-- Compiling module top

Top level modules:
        top
End time: 22:19:48 on May 27,2023, Elapsed time: 0:00:01
Errors: 0, Warnings: 0
QuestaSim-64 vopt 2022.1_1 Compiler 2022.02 Feb 19 2022
Start time: 22:19:48 on May 27,2023
vopt top -o top_opt 

Top level modules:
        top

Analyzing design...
-- Loading module top
** Error: top.sv(2): Module 'foo' is not defined.
 For instance 'u_foo' at path 'top'
Optimization failed
End time: 22:19:48 on May 27,2023, Elapsed time: 0:00:00
Errors: 1, Warnings: 0
```

(I'm not sure why the error is only reported by `vopt`, not `vlog`... but there is an error anyway.)

In contrast Verilator automatically finds `foo.sv` even if you don't list it. Note `Reading foo.sv`.

```
❯ verilator --debug --cc -sv top.sv
Starting Verilator 4.210 2021-07-07 rev v4.210
- Verilator.cpp:550:  Option --verilate: Start Verilation
- V3File.cpp:234:        --check-times failed: different command line
- V3ParseImp.cpp:271: parseFile: top
  Preprocessing top.sv
- V3PreShell.cpp:147:     Reading top.sv
- V3ParseImp.cpp:320: Lexing top.sv
- V3ParseImp.cpp:271: parseFile: foo [LIB]
  Preprocessing foo
- V3PreShell.cpp:147:     Reading foo.sv
- V3ParseImp.cpp:320: Lexing foo
dot -Tpdf -o ~/a.pdf obj_dir/Vtop_001_linkcells.dot
...
- V3Ast.cpp:1125:     Dumping obj_dir/Vtop_990_final.tree
- Verilator.cpp:732:  Done, Exiting...
```

quark17, 2023-05-27:
It sounds like the issue isn't that Verilator searches an include path, but that Verilator pre-populates the search path with the current directory.  That is, Verilator has `-y .` by default (according to the flags documentation).  Some tools don't do that and the user has to add `-y .` explicitly if they want it.  So if someone doesn't want `-y .`, is there a flag to Verilator to override it or to clear the search path?

wsnyder, 2023-05-27:
Note Questa separates parsing ("vlog") from elaboration ("vopt"), while Verilator combines those steps (unless using --hierarchical).  Even with Questa you likewise need all modules declared somehow (e.g. earlier "vlog"'s) before elaboration.

"-y ." is last in the search path so will only be used as a last resort.  So if your foo.v is in a -y you give, Verilator will find it and not use ".".

So, to summarize, either predeclare mod in any module in a filename on the command line, or have a mod.v in any -y, or as a fallback mod.v in "-y ." current directory.  There is no option to "not declare mod" as that is not legal at elaboration stage.  (Ignoring "extern module".)

wsnyder, 2023-05-27:
P.S. Any pull requests that might help improve the documentation as to what I described (or how it really works if I missed something) would be appreciated!

quark17, 2023-05-28:
I can imagine that someone might want to sanity check that they know what all the files are in a design, by explicitly listing all the files on the command line and expecting Verilator to error if a module is missing.  With the current directory automatically in the search path, they can't do that when executing Verilator in the directory with the files; they would have to move to a different directory -- which they can do, so ok there's a workaround.  But the module source files might have some include statements that pull in non-module files, and it sounds like the original poster wants those to be pulled in automatically, but still give an error for any modules not listed.  If Verilator had separate search paths for modules vs includes (and some tools do?), then that would be possible, but I think Verilator just has one search path for these?  So if the user wanted Verilator to sanity check that no module file is missing (and they had module files in the same directory with include files), the user would have to move to an empty directory, explicitly list the modules files on the command line, and also explicitly list any other include files on the command line.  (Which seems like the more likely use case.  If I wanted to confirm that I knew all the files, I'd want to confirm all the files.)

Risto97, 2023-05-28:
I also faced the issue before, I had module files in the same directory as the include files.
During verilation I had -y <that-dir> set, so when I removed the module file from the file list with CMake flag, I forgot to remove the instantiation of the component within hierarchy, done with a `ifdef, so I forgot to pass the macro.
But the verilation still worked the same way as before, as it still found the module file through -y <that-dir>.
While the other simulators were failing to find the module definition.

On the other hand changing how the modules are searched will probably break scripts for a lot of people if they were relying on this.

Timmmm, 2023-05-28:
> It sounds like the issue isn't that Verilator searches an include path, but that Verilator pre-populates the search path with the current directory.

That's a different issue I think. The issue is that it searches for modules using the include path at all. Other simulators don't. Even if I add `-incdir .` Questa won't open `foo.sv` automatically.

> On the other hand changing how the modules are searched will probably break scripts for a lot of people if they were relying on this.

I think a `--no-auto-module-search` option or similar would be fine.

quark17, 2023-05-28:
Sorry, I think understand now.  Questa has two separate search paths: the path it uses for searching for modules (which you can add to with `-y`) and the path it uses to search for included files (which you can add to with `+incdir+`).  Verilator combines these two search paths into one.  For Verilator, the flags `-y` and `+incdir+` are adding directories to the same single search path.

The Verilator behavior is clear from the Reference Guide:
```
+incdir+<dir>
    See -y.

-y <dir>
    Add the directory to the list of directories that should be
    searched to find include files or libraries.
    The three flags -y, +incdir+<dir> and -I<dir> have a similar effect;
    ...
```
However, I notice that the `--help` message is possible to interpret as separate paths:
```
+incdir+<dir>              Directory to search for includes

-y <dir>                   Directory to search for modules
```

Timmmm, 2023-05-30:
Ah yes `-y` does do it. I did try it before but didn't notice that you also need `+libext+.sv` for it to work (otherwise it just looks for a file called `foo` not `foo.sv`).

So maybe the best solution is to separate out the include paths.

wsnyder, 2023-06-04:
There was a lot of discussions of what verilator "could do".  At this time I prefer not to separate -y and +incdir+ (forcing -y only for modules and +incdir only for includes) as I think it's very rare people really want separate searches, and separating them would break existing behavior and give a bad experience to the majority and least knowledgeable users.

I'd accept a --no-module-search option if someone wants to make a pull

More complicated, if someone feels strongly that there should be a way to not find modules unless -y is given (the use case above with wanting to list every file is perhaps one), then I'd suggest a pull request along these lines:
* Make -y add a directory to a new e.g. addModDirUser()
* Make +incdir or -I add a directory to existing addIncDirUser
* When searching for files know if it is for a package/module versus an include.
* For includes, check incdir first then -y
* For modules, check -y first, then also check the incdir list.  (I was thinking -no-module-search would disable both)
* I think it falls out of above but I'd like -I to always look in both places (unless --no-module-search).

Closing assuming if that's needed a pull will come, otherwise not ;)

closed 2023-06-04 by wsnyder