verilator/verilator #4234: Unpacked structs, bug in verilated C++ for combinatorial logic from struct fields

Risto97, 2023-05-26:
@wsnyder continuation from #4226 .

Here is the sample code:

[t_unpacked_struct_comb_logic.tar.gz](https://github.com/verilator/verilator/files/11572345/t_unpacked_struct_comb_logic.tar.gz)

```systemverilog
module test (
        input wire clk,
        output logic  q
    );

    /* verilator lint_off UNOPTFLAT */
    typedef struct {
        logic overflow2;
        logic next;
    } comb_t;
    comb_t comb;
    /* verilator lint_on UNOPTFLAT */
 
    /* verilator lint_off ALWCOMBORDER */
    always_comb begin
        /* comb.next = 0; */
        if(comb.overflow2) begin 
            comb.next = 1;
        end else begin
            comb.next = 0;
        end
    end
    /* verilator lint_on ALWCOMBORDER */

    always_ff @(posedge clk) begin
        q <= comb.next;
    end
endmodule
```
And the error message:

```
Vt_unpacked_struct_comb_logic___024root__DepSet_h67077cec__0.cpp: In function ‘void Vt_unpacked_struct_comb_logic___024root___eval_triggers__act(Vt_unpacked_struct_comb_logic___024root*)’:
Vt_unpacked_struct_comb_logic___024root__DepSet_h67077cec__0.cpp:20:39: error: no match for ‘operator!=’ (operand types are ‘Vt_unpacked_struct_comb_logic_test__DOT__comb_t__struct1’ and ‘Vt_unpacked_struct_comb_logic_test__DOT__comb_t__struct1’)
   19 |     vlSelf->__VactTriggered.at(0U) = (vlSelf->test__DOT__comb
      |                                       ~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |
      |                                               Vt_unpacked_struct_comb_logic_test__DOT__comb_t__struct1
   20 |                                       != vlSelf->__Vtrigrprev__TOP__test__DOT__comb);
      |                                       ^~ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  Vt_unpacked_struct_comb_logic_test__DOT__comb_t__struct1
In file included from Vt_unpacked_struct_comb_logic__ALL.cpp:7:
Vt_unpacked_struct_comb_logic___024root__DepSet_h67077cec__0__Slow.cpp: In function ‘void Vt_unpacked_struct_comb_logic___024root___eval_triggers__stl(Vt_unpacked_struct_comb_logic___024root*)’:
Vt_unpacked_struct_comb_logic___024root__DepSet_h67077cec__0__Slow.cpp:21:39: error: no match for ‘operator!=’ (operand types are ‘Vt_unpacked_struct_comb_logic_test__DOT__comb_t__struct1’ and ‘Vt_unpacked_struct_comb_logic_test__DOT__comb_t__struct1’)
   20 |     vlSelf->__VstlTriggered.at(1U) = (vlSelf->test__DOT__comb
      |                                       ~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |
      |                                               Vt_unpacked_struct_comb_logic_test__DOT__comb_t__struct1
   21 |                                       != vlSelf->__Vtrigrprev__TOP__test__DOT__comb);
      |                                       ^~ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  Vt_unpacked_struct_comb_logic_test__DOT__comb_t__struct1
```

kbieganski, 2023-05-26:
As the message says, this breaks because the generated C++ tries to compare structs using `!=`, but this operator is not defined for those structs.

I think this works for packed structs and arrays because Verilator expands the comparison into an element-wise comparison [in V3Expand](https://github.com/verilator/verilator/blob/0ece1554180abfe8cc134bebf59f4530e6c5cc4e/src/V3Expand.cpp#L794-L815). A similar approach could be applied to unpacked structs, just instead of iterating over the width, you'd just iterate over the members and make `AstStructSel` for each one. @wsnyder what do you think?

wsnyder, 2023-05-26:
It's legal to == structs in Verilog, so perhaps jyst add the operator in the structure definitions?

kbieganski, 2023-05-26:
Ok, in that case I agree. The implementation could be quite similar to how `to_string` [is generated for classes](https://github.com/verilator/verilator/blob/0ece1554180abfe8cc134bebf59f4530e6c5cc4e/src/V3Common.cpp#L98-L150).

Risto97, 2023-05-26:
That sounds not too complicated for me to take a look.
Thank you both for the advices.

Risto97, 2023-05-27:
#4240 should fix the issue if accepted

closed 2023-05-29 by wsnyder