verilator/verilator #4235: How to remote `TOP` from output VCD file

dwilliamson, 2023-05-26:
Hello,

I'm moving away from xsim to verilator and have a big library of GTKWave `.gtkw` files for viewing traces of all my simulations. When I generate trace files both programs emit different scopes, rendering my `.gtkw` files unusable for verilator. Is there any way to fix this?

This is an example Verilog file:

```verilog
initial begin
   $dumpfile("AXILiteInterface.vcd");
   $dumpvars(0, Test_AXILiteInterface);
end
```

I run verilator with:

```bash
verilator --timing --top Test_AXILiteInterface --trace -- binary ...
```

The scope views for xsim and verilator, respectively, are:

[image]
[image]

I can obviously post-process the output as part of my build process but it would be nice if I can get verilator to behave the same as xsim.

Thanks,
\- Don

wsnyder, 2023-05-26:
TOP comes from the C++ that instantiates the Verilated model class.  Use "" if you don't want it.  (The reason for it is you can have multiple Verilated models in one simulation, unlike a "pure" Verilog simulator.)

dwilliamson, 2023-05-26:
I understand that this is the second parameter to the C++ class generated, with a default value of `"TOP"`. However, in order to change that I have to modify the call-site in the `main` function that creates it.

Is there any way I can do this from the command-line or verilog file?

wsnyder, 2023-05-26:
It's probably reasonable for the auto-generated main to default to "" to match other simulators.  Perhaps you would be willing to make a pull to add a new flag for e.g. --main-top-name that defaults to "", test it, then fix whatever tests break with that?

wsnyder, 2023-05-30:
There are many places where "" might be dropped, in the Perl wrapper, in makefiles, and elsewhere.  I'd suggest we use "-" to mean "empty" this is common with other tools e.g. "-o -".

dwilliamson, 2023-05-30:
OK, preliminary PR made. If you are OK with the way it's coded, I can add the rest of the `.rst` work.

PR #4249 (2023-05-31): Add --main-top-name option for C main TOP name (#4235)
Fixes #4235
closed 2023-05-31 by wsnyder