verilator/verilator #4237: Why the initial statement of a for loop is not seen outside of the loop?

vicencb, 2023-05-26:
Please, explain the behaviour of this code:
```verilog
module test();
  integer i;
  integer j;
  always @(i, j) $display("[%0t] B %0d %0d", $time, i, j);
  initial begin
    for(i = 1; i < 3 ; i = i + 1) begin
      for(j = 6; j < 8; j = j + 1) begin
        $display("[%0t] A %0d %0d", $time, i, j);
        #1;
        $display("[%0t] C %0d %0d", $time, i, j);
        $display("");
      end
    end
  end
endmodule
```
When this code is executed with `verilator --binary test.v && obj_dir/Vtest` it reports:
```
[0] A 1 6
[0] B 0 0
[1] C 1 6

[1] A 1 7
[1] B 0 7
[2] C 1 7

[2] A 2 6
[2] B 2 8
[3] C 2 6

[3] A 2 7
[3] B 2 7
[4] C 2 7

[4] B 3 8
```
B messages happen after A and before C. (ignoring the last one)
A and C messages match, so, i was expecting that B messages should match too.

kbieganski, 2023-05-26:
This happens because Verilator unrolls the loop and inlines all the values, so instead of actually incrementing `i` and `j` it just directly uses constant values for the `$display`s, and then sets `i` and `j` to the final values once at the end, but that happens after all the delays.

This could be fixed by disabling loop unrolling for loops containing timing controls. Putting something like
```cpp
if (nodep->exists([](AstNode* nodep) { return nodep->isTimingControl(); })) return;
```
[here in V3Unroll](
 https://github.com/antmicro/verilator-1/blob/09eaa2b7405d0dc1658a57a11eedfebca179db92/src/V3Unroll.cpp#L376) fixes your case. But there's probably a better way to do this, without additional iteration over the `while` tree. Would you be willing to try fixing this?

vicencb, 2023-05-30:
Is there an option to disable the loop unrolling optimization?
I've tried `--unroll-count 1`, but then it errors out because it can't do the optimization.

kbieganski, 2023-05-30:
That should actually work for this example. It probably errors out for you on some generate loops. I don't think there's a workaround for that, unfortunately.

wsnyder, 2023-12-24:
I committed a test for this.  Even turning off unrolling it seems there's also a scheduler issue here.