verilator/verilator #4239: How can I access the internal signals of Verilog module from C++ testbench?

always-debug, 2023-05-27:
As the title, if we have a pointer of top Verilog design (e.g dut), how the internal signals of sub-module instances are access with the pointer? I trace the C++ class which is verilated from the top Verilog module, it seems only exposing the inports and outports for access, therefore, I guess there might be derivative or compiler options for Verilator that allows accessing the design hierarchy.

Any help is very appreciated!

Thanks
/Tung

wsnyder, 2023-05-27:
https://veripool.org/guide/latest/faq.html#how-do-i-access-signals-in-c

closed 2023-05-27 by wsnyder

always-debug, 2023-05-27:
@wsnyder I have followed the document and used `/*verilator public*/` metacomment. But when I compile the Verilog module, it raises the error `VL_UNCOPYABLE(bodule_name);  ///< Copying not allowed`. Could you give me hints? I am using 4.224 version.

Thanks

wsnyder, 2023-05-27:
Probably your main is passing the model itself rather than the pointer to the module.  See the examples, and look at the line the compiler says is calling uncopyable in your code, and/or attach your code causing the error.

always-debug, 2023-05-27:
@wsnyder thanks for the hints. I fixed the compiler issue and now I can access the internal signals via metacomment. I saw the signals with metacomment are declared as CData type and when I print it via cout within the simulation loop, right after dut->eval(), I did not see any value printed out (it printed nothing). Do you know how can I resolve this?

wsnyder, 2023-05-27:
CData is a character so you need to cast to an integer to see it as a number.