verilator/verilator #4242: Non-equivalent library generation

Phantom1003, 2023-05-28:
Hi, I found that verilator incorrectly infers clock signals based on the block sensitivity list when generating libraries, which causes pure combinatorial logic libraries to have unintended behavior during simulation.

Attached is a circuit module generated by Bluespec with a sensitive list written not with * but with the use of the referenced signals.

```verilog
  always@(comb_decode_inst or
	  IF_comb_decode_inst_BITS_31_TO_26_2_EQ_0b0_3_T_ETC___d66)
  begin
    case (comb_decode_inst[14:12])
      3'b0: CASE_comb_decode_inst_BITS_14_TO_12_0b0_0_0b1__ETC__q1 = 4'd0;
      // ......
    endcase
  end
```

In the generated wrapper code you can see that these signals are identified as driving the timing circuit:
```systemverilog
    // Combinatorialy evaluate changes to inputs
    always @* begin
        last_combo_seqnum__V = controller_protectlib_combo_update(
            handle__V
            , comb_decode_combo__V
        );
    end
    
    // Evaluate clock edges
    always @(posedge comb_decode_inst or negedge comb_decode_inst) begin
        controller_protectlib_combo_ignore(
            handle__V
        );
        last_seq_seqnum__V <= controller_protectlib_seq_update(
            handle__V
            , comb_decode_inst
            , comb_decode_tmp__V
        );
        comb_decode_seq__V <= comb_decode_tmp__V;
    end
    
    // Select between combinatorial and sequential results
    always @* begin
        if (last_seq_seqnum__V > last_combo_seqnum__V) begin
            comb_decode = comb_decode_seq__V;
        end
        else begin
            comb_decode = comb_decode_combo__V;
        end
    end
```

This causes the output of the module to always be 0 during simulation.
If I change the sensitive list in the source code to *, the generated code will eliminate the incorrect sequential logic and simulate properly:

```systemverilog
    // Combinatorialy evaluate changes to inputs
    always @* begin
        last_combo_seqnum__V = controller_protectlib_combo_update(
            handle__V
            , comb_decode_inst
            , comb_decode_combo__V
        );
    end
    
    // Select between combinatorial and sequential results
    always @* begin
        comb_decode = comb_decode_combo__V;
    end
```

### Environment
- OS: Ubuntu 22.04 LTS
- Verilator: 5.002 2022-10-29 rev v5.002

[module_comb_decode.v.txt](https://github.com/verilator/verilator/files/11584163/module_comb_decode.v.txt)

wsnyder, 2023-06-04:
I'm having trouble seeing what is correct and incorrect behavior you expect in the provided test.  Can you please make this a self checking test, e.g. creates two outputs with a * and signals listed, then feeds in some stimulus and $stops if the two are not equal?

Phantom1003, 2023-07-02:
@wsnyder Sorry for the late response, I create a test case at: https://github.com/Phantom1003/verilator-issue-4242 
After executing `make`, you should get the following output:
```
[0] cyc=0       in=00000000000000000000000000000000     out_full=0000000000000000249eb6 out_star=0000000000000000000000
[0] cyc=1       in=0000000000000000000000000010041b     out_full=000000000000000021a0c8 out_star=000000000000000021a0c8
[0] cyc=2       in=00000000000000000000000000000597     out_full=000000000000000021a0c8 out_star=0000000000000000240148
[0] cyc=3       in=00000000000000000000000000000093     out_full=000000000000000021a0c8 out_star=00000000000000002100c8
[0] cyc=4       in=00000000000000000000000000000093     out_full=000000000000000021a0c8 out_star=00000000000000002100c8
```

wsnyder, 2023-07-02:
Thanks for the example I agree it mismatches; this will need further debug when someone has time - perhaps you could dig in and see why it is doing this.