verilator/verilator #4251: A few tracing questions

udif, 2023-05-30:
> How may we help - what is your question?
1. Since I control the tracing mode (--trace or --trace-fst) externally, I would like to programmatically know which tracing format is used so that I can append the correct file extension. How can I tell within the C++ code which file format is being used?
2. I've tried using FST, but if I stop my simulation with ^C (because it is stuck in an endless loop), I end up with a 330-byte waves.fst file and a larger waves.fst.hier file, and neither can be read by gtkwave. Only if my simulation ends with $finish, I get a single, large, waves.fst , which gtkwave can read.
Is there any way to flush the FST dump so I will be able to open it even if I kill the simulation in the middle?

wsnyder, 2023-06-01:
As to flushing, in Verilog use $dumpflush.  In C++ use vpi_flush(). Those should work with most all simulators.

There's also Verilator specific ways in C++, for example Verilated::runFlushCallbacks();

Closing as answered but feel free to follow up.

closed 2023-06-01 by wsnyder

udif, 2023-06-01:
Just to update, I ended up moving tracing to the C side, and will later add both VCD and FST and select between them based an environment variable or a command line flag. I hope there is no problem including and linking both, a long as I instantiate only one of the classes on each run.
Another nice feature I added was to trap SIGINT (the result of pressing ^C on the keyboard), and close the dump file before exiting.
I'm not sure it helped solving the problem 100% (I think I still see *.fst.hier files here and there), but at least it works good enough, esp. if I kill it much later than the area I wish to inspect.
This is all related to my previous DPI issues (#4225). I ended up rewriting the exported Verilog tasks as part of the DPI code. The code was cleaner and simpler before, but at least it seems to mostly work, and I got rid of the '--timing' flag.