verilator/verilator #4252: Segmentation fault @ 4f1f487ae

solomatnikov, 2023-05-31:
```
- V3Ast.cpp:1215:     Dumping /home/sols/work/wss/build/verilator_obj_dir/Vtop_006_linkjump.tree
- V3LinkInc.cpp:310:  linkIncrements: 
- V3Ast.cpp:1215:     Dumping /home/sols/work/wss/build/verilator_obj_dir/Vtop_007_linkinc.tree
- V3Param.cpp:1431:   param: 

Program received signal SIGSEGV, Segmentation fault.
AstNode::privateAs<AstNodeDType, AstNode*> (nodep=0x1) at ../V3Ast.h:2063
2063	        UASSERT_OBJ(!nodep || privateTypeTest<T>(nodep), nodep,
Missing separate debuginfos, use: yum debuginfo-install glibc-2.28-225.el8.x86_64 libgcc-8.5.0-19.el8.x86_64 libstdc++-8.5.0-19.el8.x86_64
(gdb) bt
#0  AstNode::privateAs<AstNodeDType, AstNode*> (nodep=0x1) at ../V3Ast.h:2063
#1  0x00000000005423da in AstTypedef::childDTypep (this=0x13c5a00) at ../V3Ast.h:1628
#2  AstTypedef::subDTypep (this=0x13c5a00) at ../V3AstNodeOther.h:1552
#3  AstTypedef::dump (this=0x13c5a00, str=...) at ../V3AstNodes.cpp:1761
#4  0x0000000000544490 in AstRefDType::dump (this=0x149d8c0, str=...) at ../V3AstNodes.cpp:1788
#5  0x000000000045ba14 in AstNode::v3errorEnd (this=this@entry=0x149d8c0, str=...) at ../V3Ast.cpp:1331
#6  0x000000000045bbd5 in AstNode::v3errorEndFatal (this=this@entry=0x149d8c0, str=...) at ../V3Ast.cpp:1293
#7  0x000000000057c9ca in BrokenCheckVisitor::processEnter (this=this@entry=0x7fffffffc6d0, nodep=nodep@entry=0x149d8c0) at ../V3Error.h:366
#8  0x000000000057d321 in BrokenCheckVisitor::processAndIterate (nodep=0x149d8c0, this=0x7fffffffc6d0) at ../V3Broken.cpp:313
#9  BrokenCheckVisitor::visit (this=0x7fffffffc6d0, nodep=0x149d8c0) at ../V3Broken.cpp:313
#10 0x000000000043c0aa in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:254
#11 0x000000000043c5bf in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:328
#12 0x000000000054e2ba in AstRefDType::accept (this=<optimized out>, v=...) at ../V3AstNodeDType.h:1080
#13 0x00000000004579ff in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:1000
#14 0x0000000000457a7f in AstNode::iterateChildrenConst (this=this@entry=0x149daa0, v=...) at ../V3Ast.cpp:932
#15 0x000000000057feab in VNVisitorConst::iterateChildrenConst (nodep=0x149daa0, this=0x7fffffffc6d0) at ../V3Ast.h:2512
#16 BrokenCheckVisitor::processAndIterate (nodep=0x149daa0, this=0x7fffffffc6d0) at ../V3Broken.cpp:202
#17 BrokenCheckVisitor::visit (this=0x7fffffffc6d0, nodep=0x149daa0) at ../V3Broken.cpp:305
#18 0x000000000054efe2 in AstVar::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:1763
#19 0x00000000004579ff in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:1000
#20 0x0000000000457a90 in AstNode::iterateChildrenConst (this=this@entry=0x148edf0, v=...) at ../V3Ast.cpp:933
#21 0x000000000057d32c in VNVisitorConst::iterateChildrenConst (nodep=0x148edf0, this=0x7fffffffc6d0) at ../V3Ast.h:2512
#22 BrokenCheckVisitor::processAndIterate (nodep=0x148edf0, this=0x7fffffffc6d0) at ../V3Broken.cpp:202
#23 BrokenCheckVisitor::visit (this=0x7fffffffc6d0, nodep=0x148edf0) at ../V3Broken.cpp:313
#24 0x000000000043c144 in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:263
#25 0x000000000043beb9 in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:226
#26 0x000000000054f0f8 in AstModule::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:2201
#27 0x00000000004579ff in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:1000
#28 0x0000000000457a7f in AstNode::iterateChildrenConst (this=this@entry=0x135c3a0, v=...) at ../V3Ast.cpp:932
#29 0x000000000057d32c in VNVisitorConst::iterateChildrenConst (nodep=0x135c3a0, this=0x7fffffffc6d0) at ../V3Ast.h:2512
#30 BrokenCheckVisitor::processAndIterate (nodep=0x135c3a0, this=0x7fffffffc6d0) at ../V3Broken.cpp:202
#31 BrokenCheckVisitor::visit (this=0x7fffffffc6d0, nodep=0x135c3a0) at ../V3Broken.cpp:313
#32 0x000000000043bfb2 in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:240
#33 0x000000000054ee78 in AstNetlist::accept (this=<optimized out>, v=...) at ../V3AstNodeOther.h:1153
#34 0x0000000000578d3b in VNVisitorConst::iterateConstNull (nodep=0x135c3a0, this=0x7fffffffc6d0) at ../V3Ast.h:2510
#35 BrokenCheckVisitor::BrokenCheckVisitor (nodep=0x135c3a0, this=0x7fffffffc6d0) at ../V3Broken.cpp:318
#36 V3Broken::brokenAll (nodep=nodep@entry=0x135c3a0) at ../V3Broken.cpp:348
#37 0x00000000004ebd96 in AstNode::dumpTreeFile (this=0x135c3a0, filename="/home/sols/work/wss/build/verilator_obj_dir/Vtop_008_param.tree", append=append@entry=false, 
    doDump=doDump@entry=false, doCheck=doCheck@entry=true) at ../V3Ast.cpp:1235
#38 0x000000000084afde in V3Global::dumpCheckGlobalTree (stagename="param", newNumber=newNumber@entry=0, doDump=doDump@entry=false) at ../V3Global.h:131
#39 0x00000000009df156 in V3Param::param (rootp=<optimized out>) at /usr/include/c++/8/bits/basic_string.h:256
#40 0x0000000000407e11 in process () at ../V3Global.h:131
#41 0x000000000040ab4b in verilate (
    argString="--cc --exe -sv --assert --x-assign unique --Mdir /home/sols/work/wss/build/verilator_obj_dir --top-module top +define+TESTBENCH=testbench_compute_tile +define+TESTBENCH_RAM_PATH=testbench.testmem.mem "...) at ../Verilator.cpp:636
#42 0x000000000040c24b in main (argc=<optimized out>, argv=0x7fffffffd1d8) at ../Verilator.cpp:758
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

**Not yet.**

What 'verilator --version' are you using?  Did you try it with the git master version?

**Master @ 4f1f487ae**

What OS and distribution are you using?

```
uname -a
Linux loran 4.18.0-489.el8.x86_64 #1 SMP Thu Apr 27 17:02:11 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

May we assist you in trying to fix this in Verilator yourself?

**Not sure.**

wsnyder, 2023-05-31:
Something is breaking the assertion checks on the internal tree (specifically that the tree is referencing something deleted). As you probably guessed, we'll need a test case to know what happened. If you need help to know what's wrong use --gdb and then at the prompt go "up" once get to this stack level

x00000000005423da in AstTypedef::childDTypep (this=0x13c5a00) 

then print this->fileline() to seee what line number in the sources are related to this error.

solomatnikov, 2023-05-31:
With simplified design I get:

```
%Error: Internal Error: ...: ../V3Broken.cpp:176: Broken link in node (or something without maybePointedTo): 'm_classOrPackageNodep && !m_classOrPackageNodep->brokeExists()' @ ../V3AstNodeExpr.h:736
-node: CLASSORPACKAGEREF 0x14008e0 <e2323> {h11al} @dt=0@  tl cpkg=0x13aa6d0 -> CLASS 0x13aa6d0 <e11584#> {g16ab}  tl  L5 [1ps]
   11 |    output l::l_a_ch_t l_a_channel,
      |           ^~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1215:     Dumping /home/sols/work/wss/build/verilator_obj_dir/Vtop_990_final.tree

Program received signal SIGSEGV, Segmentation fault.
AstNode::privateAs<AstNodeDType, AstNode*> (nodep=0x1) at ../V3Ast.h:2063
2063	        UASSERT_OBJ(!nodep || privateTypeTest<T>(nodep), nodep,
#0  AstNode::privateAs<AstNodeDType, AstNode*> (nodep=0x1) at ../V3Ast.h:2063
#1  0x00000000005423da in AstTypedef::childDTypep (this=0x13baed0) at ../V3Ast.h:1628
#2  AstTypedef::subDTypep (this=0x13baed0) at ../V3AstNodeOther.h:1552
#3  AstTypedef::dump (this=0x13baed0, str=...) at ../V3AstNodes.cpp:1761
#4  0x0000000000544490 in AstRefDType::dump (this=0x1440410, str=...) at ../V3AstNodes.cpp:1788
#5  0x000000000045af3c in operator<< (rhs=0x1440410, os=...) at ../V3Ast.h:2456
#6  AstNode::dumpTree (this=this@entry=0x1440410, os=..., indent="    1:2:1:", maxDepth=maxDepth@entry=-3) at ../V3Ast.cpp:1180
#7  0x000000000045b0da in AstNode::dumpTree (this=this@entry=0x14405f0, os=..., indent="    1:2:", maxDepth=maxDepth@entry=-2) at /usr/include/c++/8/ext/new_allocator.h:86
#8  0x000000000045b182 in AstNode::dumpTree (this=this@entry=0x1431320, os=..., indent="    1:", maxDepth=maxDepth@entry=-1) at /usr/include/c++/8/ext/new_allocator.h:86
#9  0x000000000045b0da in AstNode::dumpTree (this=this@entry=0x135c3a0, os=..., indent="    ", maxDepth=maxDepth@entry=0) at /usr/include/c++/8/ext/new_allocator.h:86
#10 0x00000000004ec346 in AstNode::dumpTreeFile (this=this@entry=0x135c3a0, filename="/home/sols/work/wss/build/verilator_obj_dir/Vtop_990_final.tree", append=append@entry=false, doDump=doDump@entry=true, doCheck=doCheck@entry=true) at /usr/include/c++/8/bits/char_traits.h:329
#11 0x000000000080b8be in V3ErrorGuarded::<lambda()>::operator() (__closure=0x7fffffffbcd0) at /usr/include/c++/8/ext/new_allocator.h:79
#12 std::_Function_handler<void(), V3ErrorGuarded::v3errorEnd(std::ostringstream&, const string&)::<lambda()> >::_M_invoke(const std::_Any_data &) (__functor=...) at /usr/include/c++/8/bits/std_function.h:297
#13 0x000000000040f664 in std::function<void ()>::operator()() const (this=this@entry=0x7fffffffbcd0) at /usr/include/c++/8/bits/std_function.h:260
#14 0x0000000000c0fbec in V3ThreadPool::requestExclusiveAccess(std::function<void ()> const&&) (this=this@entry=0x13473a0 <V3ThreadPool::s()::s_s>, exclusiveAccessJob=...) at ../V3ThreadPool.cpp:102
#15 0x000000000080dce0 in V3ErrorGuarded::v3errorEnd (this=this@entry=0x1347080 <V3Error::s()::s_s>, sstr=..., extra="") at ../V3ThreadPool.h:82
#16 0x000000000082e93b in FileLine::v3errorEnd (this=this@entry=0x13e6880, sstr=..., extra="") at /usr/include/c++/8/sstream:638
#17 0x000000000045bac7 in AstNode::v3errorEnd (this=this@entry=0x14008e0, str=...) at /usr/include/c++/8/bits/basic_string.h:256
#18 0x000000000045bbd5 in AstNode::v3errorEndFatal (this=this@entry=0x14008e0, str=...) at ../V3Ast.cpp:1293
#19 0x000000000057c9ca in BrokenCheckVisitor::processEnter (this=this@entry=0x7fffffffc7a0, nodep=nodep@entry=0x14008e0) at ../V3Error.h:366
#20 0x000000000057d321 in BrokenCheckVisitor::processAndIterate (nodep=0x14008e0, this=0x7fffffffc7a0) at ../V3Broken.cpp:313
#21 BrokenCheckVisitor::visit (this=0x7fffffffc7a0, nodep=0x14008e0) at ../V3Broken.cpp:313
#22 0x000000000043c0de in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:257
#23 0x000000000043b405 in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:72
#24 0x000000000054e642 in AstClassOrPackageRef::accept (this=<optimized out>, v=...) at ../V3AstNodeExpr.h:733
#25 0x00000000004579ff in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:1000
#26 0x0000000000457a90 in AstNode::iterateChildrenConst (this=this@entry=0x14007e0, v=...) at ../V3Ast.cpp:933
#27 0x000000000057d32c in VNVisitorConst::iterateChildrenConst (nodep=0x14007e0, this=0x7fffffffc7a0) at ../V3Ast.h:2512
#28 BrokenCheckVisitor::processAndIterate (nodep=0x14007e0, this=0x7fffffffc7a0) at ../V3Broken.cpp:202
#29 BrokenCheckVisitor::visit (this=0x7fffffffc7a0, nodep=0x14007e0) at ../V3Broken.cpp:313
#30 0x000000000043c0aa in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:254
#31 0x000000000043c5bf in VNVisitorConst::visit (this=<optimized out>, nodep=<optimized out>) at ./V3Ast__gen_visitor_defns.h:328
#32 0x000000000054e2ba in AstRefDType::accept (this=<optimized out>, v=...) at ../V3AstNodeDType.h:1080
#33 0x00000000004579ff in AstNode::iterateAndNextConst (this=<optimized out>, v=...) at ../V3Ast.cpp:1000
#34 0x0000000000457a7f in AstNode::iterateChildrenConst (this=this@entry=0x14009c0, v=...) at ../V3Ast.cpp:932
#35 0x000000000057feab in VNVisitorConst::iterateChildrenConst (nodep=0x14009c0, this=0x7fffffffc7a0) at ../V3Ast.h:2512
#36 BrokenCheckVisitor::processAndIterate (nodep=0x14009c0, this=0x7fffffffc7a0) at ../V3Broken.cpp:202
[... cut ...]
#59 0x00000000009df156 in V3Param::param (rootp=<optimized out>) at /usr/include/c++/8/bits/basic_string.h:256
#60 0x0000000000407e11 in process () at ../V3Global.h:131
#61 0x000000000040ab4b in verilate (argString="--cc --exe -sv --assert --x-assign unique --Mdir /home/sols/work/wss/build/verilator_obj_dir --top-module top +define+TESTBENCH=testbench_compute_tile +define+TESTBENCH_RAM_PATH=testbench.testmem.mem "...) at ../Verilator.cpp:636
#62 0x000000000040c24b in main (argc=<optimized out>, argv=0x7fffffffd2a8) at ../Verilator.cpp:758
```

solomatnikov, 2023-05-31:
Test case:

```
git diff
diff --git a/test_regress/t/t_class_typedef.v b/test_regress/t/t_class_typedef.v
index e5034a601..836e0c088 100644
--- a/test_regress/t/t_class_typedef.v
+++ b/test_regress/t/t_class_typedef.v
@@ -12,6 +12,24 @@ class uvm_resource_pool;
    uvm_resource_types::rsrc_q_t rtab [string];
 endclass
 
+class B#(parameter SIZE = 2);
+    typedef logic [SIZE-1:0] t_vector;
+    typedef struct {
+        t_vector m0;
+    } t_struct;
+endclass
+
+module t_io_class_typedef (
+   // Inputs
+   input  B::t_struct in_struct,
+   // Outputs
+   output B::t_vector out_vector
+   );
+
+   assign out_vector = in_struct.m0;
+
+endmodule
+
 virtual class C#(parameter type T = logic, parameter SIZE = 1);
     typedef logic [SIZE-1:0] t_vector;
     typedef T t_array [SIZE-1:0];
@@ -22,6 +40,13 @@ virtual class C#(parameter type T = logic, parameter SIZE = 1);
 endclass
 
 module t (/*AUTOARG*/);
+
+   B::t_struct b_struct;
+   B::t_vector b_vector;
+
+   t_io_class_typedef i_t_io_class_typedef(.in_struct (b_struct),
+                                          .out_vector(b_vector));
+
    initial begin
       uvm_resource_pool pool = new;
       typedef logic [7:0] t_t0;
```

Internal Error: 

```
test_regress/t/t_class_typedef.pl
======================================================================
dist/t_class_typedef: ==================================================
-Skip: dist/t_class_typedef: scenario 'dist' not enabled for test
dist/t_class_typedef: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Time 0:00
======================================================================
vlt/t_class_typedef: ==================================================
	perl /home/sols/work/verilator/test_regress/../bin/verilator --prefix Vt_class_typedef ../obj_vlt/t_class_typedef/Vt_class_typedef__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_class_typedef --fdedup --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_class_typedef +define+TEST_DUMPFILE=obj_vlt/t_class_typedef/simx.vcd t/t_class_typedef.v    > obj_vlt/t_class_typedef/vlt_compile.log
%Error: Internal Error: t/t_class_typedef.v:44:7: ../V3Broken.cpp:176: Broken link in node (or something without maybePointedTo): 'm_typedefp && !m_typedefp->brokeExists()' @ ../V3AstNodes.cpp:1803
                                                : ... In instance t
   44 |    B::t_struct b_struct;
      |       ^~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
%Warning: vlt/t_class_typedef: Exec of perl failed: %Error: Internal Error: t/t_class_typedef.v:44:7: ../V3Broken.cpp:176: Broken link in node (or something without maybePointedTo): 'm_typedefp && !m_typedefp->brokeExists()' @ ../V3AstNodes.cpp:1803

vlt/t_class_typedef: %Error: Exec of perl failed: %Error: Internal Error: t/t_class_typedef.v:44:7: ../V3Broken.cpp:176: Broken link in node (or something without maybePointedTo): 'm_typedefp && !m_typedefp->brokeExists()' @ ../V3AstNodes.cpp:1803
vlt/t_class_typedef: FAILED: Exec of perl failed: %Error: Internal Error: t/t_class_typedef.v:44:7: ../V3Broken.cpp:176: Broken link in node (or something without maybePointedTo): 'm_typedefp && !m_typedefp->brokeExists()' @ ../V3AstNodes.cpp:1803
==SUMMARY: Passed 0  Failed 1  Time 0:00
==SUMMARY: Passed 0  Failed 1  Time 0:00

======================================================================
	#vlt/t_class_typedef: %Error: Exec of perl failed: %Error: Internal Error: t/t_class_typedef.v:44:7: ../V3Broken.cpp:176: Broken link in node (or something without maybePointedTo): 'm_typedefp && !m_typedefp->brokeExists()' @ ../V3AstNodes.cpp:1803
		make && test_regress/t/t_class_typedef.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Time 0:00
```

wsnyder, 2023-06-01:
Looks like it's getting confused by the B:: references, probably because the class is parameterized and used as a parameterized class.  Perhaps you could look into this and provide a pull to fix it?

solomatnikov, 2023-06-01:
What does it involve?

I tried to go up the stack in gdb but didn't see anything obvious.

Also, it looks like there are 2 separate issues:

1. Internal error.
2. Seg. fault in debug mode when trying to dump the tree.

solomatnikov, 2023-06-01:
Maybe #3809 has to be fixed first?

solomatnikov, 2023-06-01:
This passed:

```
git diff
diff --git a/test_regress/t/t_class_typedef.v b/test_regress/t/t_class_typedef.v
index e5034a601..e06d974fe 100644
--- a/test_regress/t/t_class_typedef.v
+++ b/test_regress/t/t_class_typedef.v
@@ -12,6 +12,24 @@ class uvm_resource_pool;
    uvm_resource_types::rsrc_q_t rtab [string];
 endclass
 
+class B#(parameter SIZE = 2);
+    typedef logic [SIZE-1:0] t_vector;
+    typedef struct {
+        t_vector m0;
+    } t_struct;
+endclass
+
+module t_io_class_typedef (
+   // Inputs
+   input  B::t_struct in_struct,
+   // Outputs
+   output B::t_vector out_vector
+   );
+
+   assign out_vector = in_struct.m0 + 2'd1;
+
+endmodule
+
 virtual class C#(parameter type T = logic, parameter SIZE = 1);
     typedef logic [SIZE-1:0] t_vector;
     typedef T t_array [SIZE-1:0];
@@ -22,6 +40,15 @@ virtual class C#(parameter type T = logic, parameter SIZE = 1);
 endclass
 
 module t (/*AUTOARG*/);
+
+   B#()::t_struct b_struct;
+   B#()::t_vector b_vector;
+
+   assign b_struct.m0 = 2'd1;
+
+   t_io_class_typedef i_t_io_class_typedef(.in_struct (b_struct),
+                                          .out_vector(b_vector));
+
    initial begin
       uvm_resource_pool pool = new;
       typedef logic [7:0] t_t0;
```

solomatnikov, 2023-06-08:
Replaced by #4280, #4281

closed 2023-06-08 by solomatnikov