verilator/verilator #4255: Native Windows support

Timmmm, 2023-05-31:
I'm trying to run Verilator on Windows. As far as I can see there are a number of suboptimal options:

* Compile it myself using MSYS2/Cygwin/Mingw etc. These are all a pain to set up and suck in various ways.
* Run it under WSL. Not really Windows and this introduces other hassles.
* The only precompiled binary I could find is in [this package](https://github.com/YosysHQ/oss-cad-suite-build), unfortunately it doesn't seem to work (it looks for `verilated_std.sv` in a location that doesn't exist.

It would be nice if you could download and use Verilator in the same way you can use most Windows software - even software with questionable distribution practices like Python.

I think at a minimum that means:

1. Ditching the `verilator` Perl script. Windows users do not have Perl. Probably the easiest option is to rewrite it in C++. Or maybe just get rid of it entirely? Most programs don't have a script frontend to find themselves and prepend `gdb` etc.
2. Rework the `VERILATOR_ROOT` mechanism. Verilator should look for its files relative to where `verilator_bin.exe` is.
3. Build Windows binaries in CI and put them in Github Releases.

wsnyder, 2023-06-01:
At this time I am reluctant to add a native Windows build because that adds an extreme support cost I think it is unlikely can be sustained.  This will likely improve over time - for example if you or others help out ;)

However, discussing the issues you point out and patches to solve them would be good steps forward.

I do not want to lose the verilator perl script's functions as this is massively useful in supporting problems.  Creating a C++ wrapper that does similar functions would be fine.  I've also been working over time in removing perl entirely so that would help that effort.  (For various reasons just replacing perl with Python wasn't desirable.)

Adding additional VERILATOR_ROOT replacements is possible, I've been reluctant to add yet more install options, but perhaps this is a case where we have little choice.  We'll need to debate what's "right" layout wise - perhaps just default VERILATOR_ROOT to "(pwd)/.." if not set and not compiled in.

Please file an issue on whichever (or neither) you are willing to work on.

kbieganski, 2023-06-01:
> We'll need to debate what's "right" layout wise - perhaps just default VERILATOR_ROOT to "(pwd)/.." if not set and not compiled in.

I think defaulting to the parent dir of `verilator_bin` is a good option.

- It would work on Windows, where all the app-related files are stored in the same root directory as the binary.
- It would be useful for development – when I work on Verilator, I often have a few different builds to compare, and not having to set `VERILATOR_ROOT` for those is more convenient and less error-prone.

wsnyder, 2024-03-26:
I don't know what you are asking to do, please clarify.

pidgeon777, 2024-03-26:
I see that *verilator* has to be called by executing `verilator`, which is a Perl script.

In my case it is found in:

`C:\Users\<USERNAME>\scoop\apps\oss-cad-suite-nightly\current\bin\verilator`

The `verilator` script also calls `verilator_bin.exe`, which is a Windows executable.

First I was wondering if it is "allowed" to call `verilator_bin.exe` directly, without calling the Perl script, or if there are any reasons why this is not recommended.

If it's allowed, why is the Perl script used at all? Is it just for convenience, or are there any other reasons?

Also, the Perl script seems to be compatible with Linux only, I had to make some changes to get it running on Windows (e.g. changing the path separators from `/` to `\`).

Moreover, if no `--no-unlimited-stack` Perl script argument is passed, the script will attempt to run a system command that is not available on Windows:

`system("ulimit -s unlimited 2>/dev/null");`

So, when running `verilator` on Windows, it is necessary to pass the `--no-unlimited-stack` argument:

`perl verilator --no-unlimited-stack ...`

Finally, it seems that the `VERILATOR_ROOT` environment variable is used only by the Perl script. So, it wouldn't be necessary to set it if the Windows executable is called directly.

wsnyder, 2024-03-26:
@pidgeon777, as you found the script is there to run certain setup steps before the binary. This is always expected to be run first; bypassing it may just cause later problems downstream when the e.g. includes are not found; VERILATOR_ROOT is used by the binary.  However the master (unreleased) is better at finding the binary without VERILATOR_ROOT set (#4927).

As you also found we do not support native windows paths, hence this bug.

I encourage use of WSL instead of windows, or native Linux, not just because of this bug, but as EDA chip industry usage generally uses Linux.

wsnyder, 2024-04-13:
Unlikely to support windows native any time soon so just closing.

closed 2024-04-13 by wsnyder