verilator/verilator #4256: C++ compile error @ 4f1f487ae

solomatnikov, 2023-06-01:
```
g++ -O0  -I.  -MMD -I.../verilator/test_regress/../include -I.../verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_class_typedef -DVM_PREFIX=Vt_class_typedef -DVM_PREFIX_INCLUDE="<Vt_class_typedef.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_class_typedef___024root.h>" -DT_CLASS_TYPEDEF   -c -o Vt_class_typedef__ALL.o Vt_class_typedef__ALL.cpp
In file included from Vt_class_typedef___024root.h:10,
                 from Vt_class_typedef__Syms.h:17,
                 from Vt_class_typedef.cpp:5,
                 from Vt_class_typedef__ALL.cpp:3:
Vt_class_typedef___024unit__03a__03aB__W3__Vclpkg.h:15:8: error: redefinition of ‘struct Vt_class_typedef_t_struct__struct3’
 struct Vt_class_typedef_t_struct__struct3 {
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Vt_class_typedef___024root.h:9,
                 from Vt_class_typedef__Syms.h:17,
                 from Vt_class_typedef.cpp:5,
                 from Vt_class_typedef__ALL.cpp:3:
Vt_class_typedef___024unit__03a__03aB__Vclpkg.h:15:8: note: previous definition of ‘struct Vt_class_typedef_t_struct__struct3’
 struct Vt_class_typedef_t_struct__struct3 {
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Vt_class_typedef__ALL.cpp:13:
Vt_class_typedef___024unit__03a__03aB__W3__Vclpkg__DepSet_hf3df3bd9__0.cpp:10:13: error: redefinition of ‘std::__cxx11::string VL_TO_STRING(const Vt_class_typedef_t_struct__struct3&)’
 std::string VL_TO_STRING(const Vt_class_typedef_t_struct__struct3& obj) {
             ^~~~~~~~~~~~
In file included from Vt_class_typedef__ALL.cpp:11:
Vt_class_typedef___024unit__03a__03aB__Vclpkg__DepSet_hadfb9064__0.cpp:10:13: note: ‘std::__cxx11::string VL_TO_STRING(const Vt_class_typedef_t_struct__struct3&)’ previously defined here
 std::string VL_TO_STRING(const Vt_class_typedef_t_struct__struct3& obj) {
             ^~~~~~~~~~~~
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
class uvm_resource_types;
   typedef int rsrc_q_t;
endclass

class uvm_resource_pool;
   uvm_resource_types::rsrc_q_t rtab [string];
endclass

class B#(parameter WIDTH = 2);
    typedef logic [WIDTH-1:0] t_vector;
    typedef struct {
        t_vector m0;
    } t_struct;
endclass

module t_io_class_typedef   #(parameter WIDTH = 2) (
   // Inputs
   input  B#(WIDTH)::t_struct in_struct,
   // Outputs
   output B#(WIDTH)::t_vector out_vector
   );

   assign out_vector = in_struct.m0 + 2'd1;

endmodule

virtual class C#(parameter type T = logic, parameter SIZE = 1);
    typedef logic [SIZE-1:0] t_vector;
    typedef T t_array [SIZE-1:0];
    typedef struct {
        t_vector m0 [2*SIZE-1:0];
        t_array m1;
    } t_struct;
endclass

module t (/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   B#()::t_struct b_struct1;
   B#()::t_vector b_vector1;

   B#(3)::t_struct b_struct2;
   B#(3)::t_vector b_vector2;

   assign b_struct1.m0 = 2'd1;
   assign b_struct2.m0 = 3'd2;

   t_io_class_typedef i_t_io_class_typedef1(.in_struct (b_struct1),
					    .out_vector(b_vector1));

   t_io_class_typedef #(3) i_t_io_class_typedef2 (.in_struct (b_struct2),
					          .out_vector(b_vector2));

   initial begin
      uvm_resource_pool pool = new;
      typedef logic [7:0] t_t0;
      C#(t_t0,3)::t_vector v0;
      C#(t_t0,3)::t_array a0;
      C#(bit,4)::t_struct s0;

      pool.rtab["a"] = 1;
      if ($bits(pool.rtab["a"]) != 32) $stop;

      if ($bits(v0) != 3) $stop;
      if ($size(a0) != 3) $stop;
      if ($bits(a0[0]) != 8) $stop;
      if ($size(s0.m0) != 8) $stop;
      if ($size(s0.m1) != 4) $stop;
      if ($bits(s0.m1[2]) != 1) $stop;
      if ($bits(b_struct1.m0) != 2) $stop;
      if ($bits(b_vector1) != 2) $stop;
      if ($bits(b_struct2.m0) != 3) $stop;
      if ($bits(b_vector2) != 3) $stop;
   end

   always @ (posedge clk) begin
      if (b_vector1 != 2'd2) $stop;
      if (b_vector2 != 3'd3) $stop;

      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule
```

What 'verilator --version' are you using?  Did you try it with the git master version?

**Master @ https://github.com/verilator/verilator/commit/4f1f487ae4ea0811883e78d287941a4d8abbad19**

What OS and distribution are you using?

```
uname -a
Linux loran 4.18.0-492.el8.x86_64 #1 SMP Tue May 9 17:56:55 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

May we assist you in trying to fix this in Verilator yourself?

**Not sure**

solomatnikov, 2023-06-01:
However the test works if 1 line is commented out:
```
//      if (b_vector2 != 3'd3) $stop;
```

Risto97, 2023-06-01:
https://github.com/verilator/verilator/issues/4257#issuecomment-1572414099

I can try to guess.
It looks like that the struct gets defined twice in the Vt_class_typedef___024root.h.
Probably because you have parameters and verilator does not understand that it already defined a struct with the given parameter 3, so it defines it twice.

Keep in mind I am also new to the verilator codebase, so I am not sure if what I guessed is correct.
My advice is to try and create a minimum possible example to reproduce the bug, its usually much easier to track the problem.

Risto97, 2023-06-01:
I think the solution would be when you have a parametrized class, is that all the members should be either in a new namespace containing class name and normalized parameters or the members should include the normalized parameter in the name.

kiryk, 2023-06-02:
It's unexpected because normally the C++ code of every struct is emitted only once, into a header file. Then if it's used in multiple modules/files each of them just includes that header.

I've noticed that in your example, after Verilator finishes elaboration, the symbol table contains two separate but identical struct definitions. Since this can happen, normally the name of each struct is appended with a unique number to avoid redefinition errors. But these two somehow get the same number, which could be because they originate from the same generic type.

So I can see two things that could cause this bug:
1. Verilator should notice that these two structs are the same, and make them into one.
2. The unique number should be calculated in a different way.

I'd opt for 2., because if I interpret the IEEE standard correctly (section 6.22.1 on type equivalence) those two structs should be treated as different types. If so, then this is a simple fix. I pushed it [here](https://github.com/antmicro/verilator-1/tree/kiryk/fix-struct-redefinition), and I'll probably open a PR soon.

I tested it with your example, and it seems to work.

wsnyder, 2023-06-03:
I agree IEEE requires them to be separate for checking type equivalence, but once V3Width has finished elaboration and checking data types that should not matter any more.  Verilator could then combine equivalent structs, which potentially would make for less & faster code.

However Verilator doesn't generally combine types currently, so I'm fine proceeding with making them uniquely named for now.

solomatnikov, 2023-06-03:
> the symbol table contains two separate but identical struct definitions. Since this can happen, normally the name of each struct is appended with a unique number to avoid redefinition errors. But [...]

The intent of the test is to have 2 different structs: one with default class params and another with non-default class params.

If the structs end up the same, it may be another bug in class parameterization.

PR #4276 (2023-06-06): Fix struct redefinition
This fixes #4256 by changing the way names of unpacked structs are made unique.

It seems that cloneTree copies uniqueNum of the original type, so generic types can have conflicting numbers.
I think a general fix would require custom copy constructors for type classes.
closed 2023-06-06 by wsnyder