verilator/verilator #4257: C++ compile error no match for ‘operator==’ @ 4f1f487ae

solomatnikov, 2023-06-01:
```
g++ -O0  -I.  -MMD -I.../verilator/test_regress/../include -I.../verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_class_typedef -DVM_PREFIX=Vt_class_typedef -DVM_PREFIX_INCLUDE="<Vt_class_typedef.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_class_typedef___024root.h>" -DT_CLASS_TYPEDEF   -c -o Vt_class_typedef__ALL.o Vt_class_typedef__ALL.cpp
In file included from Vt_class_typedef___024root.h:10,
                 from Vt_class_typedef__Syms.h:17,
                 from Vt_class_typedef.cpp:5,
                 from Vt_class_typedef__ALL.cpp:3:
Vt_class_typedef___024unit__03a__03aC__Vclpkg.h: In member function ‘bool Vt_class_typedef_t_struct__struct5::operator==(const Vt_class_typedef_t_struct__struct5&)’:
Vt_class_typedef___024unit__03a__03aC__Vclpkg.h:22:26: error: no match for ‘operator==’ (operand types are ‘VlUnpacked<unsigned char, 1>’ and ‘const VlUnpacked<unsigned char, 1>’)
             && __PVT__m1 == rhs.__PVT__m1;
                ~~~~~~~~~~^~~~~~~~~~~~~~~~
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
class uvm_resource_types;
   typedef int rsrc_q_t;
endclass

class uvm_resource_pool;
   uvm_resource_types::rsrc_q_t rtab [string];
endclass

class B#(parameter WIDTH = 2);
    typedef logic [WIDTH-1:0] t_vector;
    typedef struct {
        t_vector m0;
    } t_struct;
endclass

module t_io_class_typedef #(parameter WIDTH = 2) (
   // Inputs
   input  B#(WIDTH)::t_struct in_struct,
   // Outputs
   output B#(WIDTH)::t_vector out_vector
   );

   assign out_vector = in_struct.m0 + 2'd1;

endmodule

virtual class C#(parameter type T = logic, parameter SIZE = 1);
    typedef logic [SIZE-1:0] t_vector;
    typedef T t_array [SIZE-1:0];
    typedef struct {
        t_vector m0 [2*SIZE-1:0];
        t_array m1;
    } t_struct;
endclass

module t_io_vclass_typedef (
   // Inputs
   input  C::t_struct in_struct,
   // Outputs
   output C::t_vector out_vector,
   output C::t_array  out_array
   );

   assign out_vector = in_struct.m0[0] + in_struct.m0[1];
//   assign out_array  = in_struct.m1;

endmodule

module t (/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;

   B#()::t_struct b_struct1;
   B#()::t_vector b_vector1;

   B#(3)::t_struct b_struct2;
   B#(3)::t_vector b_vector2;

   C#()::t_struct c_struct1;
   C#()::t_vector c_vector1;
   C#()::t_array  c_array1;

   assign b_struct1.m0 = 2'd1;
   assign b_struct2.m0 = 3'd2;

   assign c_struct1.m0[0] = 1'd0;
   assign c_struct1.m0[1] = 1'd1;
   assign c_struct1.m1[0] = 1'd1;

   t_io_class_typedef i_t_io_class_typedef1(.in_struct (b_struct1),
					    .out_vector(b_vector1));

   t_io_class_typedef #(3) i_t_io_class_typedef2 (.in_struct (b_struct2),
					          .out_vector(b_vector2));

   t_io_vclass_typedef i_t_io_vclass_typedef1(.in_struct (c_struct1),
				              .out_vector(c_vector1),
					      .out_array (c_array1));

   initial begin
      uvm_resource_pool pool = new;
      typedef logic [7:0] t_t0;
      C#(t_t0,3)::t_vector v0;
      C#(t_t0,3)::t_array a0;
      C#(bit,4)::t_struct s0;

      pool.rtab["a"] = 1;
      if ($bits(pool.rtab["a"]) != 32) $stop;

      if ($bits(v0) != 3) $stop;
      if ($size(a0) != 3) $stop;
      if ($bits(a0[0]) != 8) $stop;
      if ($size(s0.m0) != 8) $stop;
      if ($size(s0.m1) != 4) $stop;
      if ($bits(s0.m1[2]) != 1) $stop;
      if ($bits(b_struct1.m0) != 2) $stop;
      if ($bits(b_vector1) != 2) $stop;
      if ($bits(b_struct2.m0) != 3) $stop;
      if ($bits(b_vector2) != 3) $stop;
      if ($bits(c_vector1) != 1) $stop;
   end

   always @ (posedge clk) begin
      if (b_vector1 != 2'd2) $stop;
//      if (b_vector2 != 3'd3) $stop;
      if (c_vector1 != 1'd1) $stop;

      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule
```

What 'verilator --version' are you using?  Did you try it with the git master version?

**Master @ https://github.com/verilator/verilator/commit/4f1f487ae4ea0811883e78d287941a4d8abbad19**

What OS and distribution are you using?

```
uname -a
Linux loran 4.18.0-492.el8.x86_64 #1 SMP Tue May 9 17:56:55 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

May we assist you in trying to fix this in Verilator yourself?

**Not sure**

solomatnikov, 2023-06-01:
However the test works if one line is commented out:
```
//      if (c_vector1 != 1'd1) $stop;
```

Risto97, 2023-06-01:
Can you try with this commit? https://github.com/verilator/verilator/commit/7f471d862e47895303378b94807021a074b49ad1
If it doesn't fix it let me know, because I was just working on it.

closed 2023-06-01 by solomatnikov

solomatnikov, 2023-06-01:
@Risto97 btw, do you have any ideas about what's going on in #4256 ?