verilator/verilator #4259: Verilator Code flow

UET76, 2023-06-01:
Hi! all. I wanted to understand verialtor code flow,how it generates ASTs and all. For example consider one simple line of code: 
assign y = a&b. so how does verilator generates a c++ output. i did dumped a number of AST and then i visualized them but i can't understand all that code flow. if anyone could guide
![Vand_001_cells tree](https://github.com/verilator/verilator/assets/135226
[image]
376/e8066926-a19c-4d13-b717-6dbd06767064)

closed 2023-06-01 by wsnyder

UET76, 2023-06-01:
that was harsh coming from people like like you. I always thought there was no harm in asking for help but forgot that no harm in asking the right person for help

Engrr-Dawood, 2023-06-01:
Sir, I've read that PDF a couple of times. My Supervisor asked me to explain how veriltor generates an AST since it's a part of my research now. I did run the verilator in debug mod and using the -dump-dot command and then visualized those tree files with Graphviz. But after that i am stuck. Little I know that it identifies the root node and then traverses downwards creating children nodes

udif, 2023-06-01:
You should familiarize yourself with the concepts of parsers and lexers.
Verilator is using two very common tools of the C/Unix echosystem called Bison and Flex, that build on older tools called YACC and LEX.
Try searching google for tutorials about them. Useful search terms might be "bison parser generator" and "flex lexical analyzer"

Another useful way to learn about this is to enroll to an online compilers course such as:
https://www.edx.org/course/compilers