verilator/verilator #4263: Add support for ## cycle delays

amykyta3, 2023-06-02:
Currently Verilator does not support `##N` cycle delays.
These are essential counterparts to clocking blocks, especially the special case of a `##0` synchronization delay as it enables one to design transactor agents that support proper back-to-back transfers.

wsnyder, 2023-06-03:
Note $past is supported which is somewhat similar.

A pull request to well self-test ##, and then to ideally implement it, would be appreciated.

kbieganski, 2023-06-07:
I would say it does support cycle delays, see the [t_clocking_timing](https://github.com/verilator/verilator/blob/master/test_regress/t/t_clocking_timing.v#L55-L89) test. Bugs are likely though. Did you try using them? Is there a more specific problem?

jscheid-ventana, 2023-06-08:
The particulars of `##0` would need to be covered, particularly in procedural context.  From 1800 "14.11 Cycle delay: ##".

closed 2026-03-20 by wsnyder