verilator/verilator #4273: Question: How to compile a .so file?

davidbai13, 2023-06-06:
I want to compile the system verilog file into a .so file.
(ex system verilog -> /cpp's -> . /obj_dir/VTop.so )
Or select the files from the generated cpp's that can be used to generate so files and then use gcc to generate the so files.
Is there a way to do this please?
thank you

wsnyder, 2023-06-06:
Note C++ may behave badly if a shared object that contains C++ standard library calls (like Verilator uses) isn't loaded into a main program that used the exact same version of C++ standard library.   So don't do this if you can't ensure this.

Verilator doesn't have rules to make .so files itself.  So you'll need to write your own makefile rule to make the shared object.

You can see all of the .cpp files needed in the generated .mk file.  Then you'll need to figure out the right $(LD) command to build a shared object, modeling on the default rule in verilated.mk, e.g. something like (untested)

$(VM_PREFIX)__ALL.so: $(VK_OBJS) $(VM_HIER_LIBS)
   $(LD) $(LDFLAGS) $(LOADLIBES) $(LDLIBS) -shared -o $@ $^

You'll have to experiment with this, shared objects can be complicated to get right.

When you get a rule/process that works here please report back to help the next person.

wsnyder, 2023-06-17:
Closing as answered (with ideas as opposed to exact recipe) but please feel free to follow up with how you did it, etc.

closed 2023-06-17 by wsnyder