verilator/verilator #4278: How to properly create a working bidir pad for use with verilator.

phelter, 2023-06-07:
I have a bidirectional pad model:

```verilog
module pad_model_bd ( 
    output logic RX_OUT, //% digital side Receive output <- input from PAD
    input wire TX_IN,  //% digital side Transmit value -> output to PAD
    input wire RXEN,  //% digital side Receive enable
    input wire TXEN,  //% digital side Transmit enable

    inout wire PAD, //% Pad connection

    inout wire PWROK,
    // verilator lint_off UNUSEDSIGNAL
    inout wire VDDIO,
    inout wire VDD,
    inout wire VSSIO,
    inout wire VSS
    // verilator lint_on UNUSEDSIGNAL
);

    // Require access to these variables for testing output
    logic PAD_txen;
    logic PAD_txout;
    always_comb begin
        PAD_txen  = (TXEN & PWROK);
        PAD_txout = (PAD_txen) ? TX_IN : 1'b0;
    end

    // Tried this but it doesn't allow a value to be set to pad i.e. m_dut->PAD = 0 doesn't propagate a value on PAD -> RXOUT
    // always PAD = (PAD_txen) ? TX_IN : 1'bz;

    // Input logic
    always_comb begin
        if (RXEN & PWROK) begin
            RX_OUT = PAD;
        end else begin
            RX_OUT = 1'b0;
        end
    end

    // Workaround to allow me to access what the pad output would be.
    export "DPI-C" task pad_bd_getPadOut;

    task pad_bd_getPadOut(output bit value, output bit enable);
        enable = PAD_txen;
        value  = PAD_txout;
    endtask

endmodule
```

So using this model - I can always access what would have been placed on the PAD output via the `pad_bd_getPadOut` DPI task - yes I know all the issues related to accessing this and not knowing when, but I have a scenario where the testbench in C++ land is the controller of sequencing, so I can access the task at known times (i.e. when things are settled and not at an edge).

However when I wrap this within a couple levels of hierarchy- note the example level of hierarchy below is verilog and not SystemVerilog and are written in this manner (not identifying wires/logic/regs because they are the models from a third-party vendor that I'm trying to mimic for behavioral sims in verilator).

```verilog
module wrapper_pad_bd ( RX_OUT,  PAD, DRV, TX_IN, RXEN, TXEN, PWROK, VDDIO, VDD, VSSIO, VSS );
output RX_OUT;
input TX_IN;
input RXEN;
input TXEN;

inout PAD;
inout PWROK;

inout VDDIO;
inout VDD;
inout VSSIO;
inout VSS;

    pad_model_bd verilator_model (
       .RX_OUT(RX_OUT),
       .TX_IN(TX_IN),
       .RXEN(RXEN),
       .TXEN(TXEN),
       .PAD(PAD),
       .PWROK(PWROK),
       .VDDIO(VDDIO),
       .VDD(VDD),
       .VSSIO(VSSIO),
       .VSS(VSS)
    );

endmodule
```

The PAD port when setting a value on it via C++  eg `m_dut->PAD = 1` - does not propagate to the lower level of hierarchy.

So I'm wondering what fundamental thing am I missing?

Example of pad_model_bd - checking RX_OUT logic (no wrapper):  Eg TOP->pad_model_bd
[image]

Example of pad_model_bd - checking RX_OUT logic (no wrapper - but with `always PAD = (PAD_txen) ? TX_IN : 1'bz;` un-commented).
[image]
Notice PAD never gets the value pushed via a `m_dut->PAD = 0 / 1` from the C++ testbench.

So a similar thing happens when I have multiple levels of hierarchy as well.  The toplevel net/ port name has a value changing but all layers of hierarchy do not.

I'm at a point where I think I'll have to make a DPI-C function for the input PAD -> RX_OUT as well at this point.  And forget checking the connectivity of the very top to the PAD inout value.

Appreciate the help.

phelter, 2023-06-09:
I was thinking you might have a working example of a pad to mimic that might be in your suite of tests for verilator.

Eg;
```verilog
module pad_bidir ( 
    output logic rx_out, //% Digital side receive output <- PAD
    input wire tx_in,    //% Digital side transmit input -> PAD
    input wire rx_en,    //% Digital side receive enable
    input wire tx_en,    //% Digital side transmit enable

    inout wire PAD
);

assign rx_out = PAD & rx_en;
assign PAD = tx_en ? tx_in : 1'bz;

endmodule
```

And an example of how to read the value at the PAD and set the value of the pad via C++  so it affects the rx_out signal.

I'm missing what I need to do there for the verilated C++ code.

The other scenario I have which may be simpler but shows full bidir connectivity is just a wire:

```verilog
module pad_wire ( 
    inout wire ay, //% Digital side inout wire.
    inout wire PAD //% Pad side inout wire.
);
    assign PAD = ay;
endmodule

```

What I would like is a working example of how to set the PAD value and have the corresponding ay signal change to the value of PAD.

eg psuedo code of test.
```cpp
   m_dut->PAD = 0;
   tick(); //< increment by smallest unit of time and eval() and write out traces.
   EXPECT_THAT(m_dut->ay, Eq(0));

   m_dut->PAD = 1;
   tick(); //< increment by smallest unit of time and eval() and write out traces.
   EXPECT_THAT(m_dut->ay, Eq(1));
```
Right now when I change the value of PAD - there is no change to the value in the waves and nothing propagates.  So how is it that you use an `inout` port in the C++ land?

And then vice-versa a working example of how to set the ay value and have the corresponding PAD wire change to the value of ay.  (assuming this will be the same).

I have yet to find a good example of how this is done on github or anywhere else - with verilator usage.

I did look at the `generated_tristate.tree` - for both of these, but when I look at the instance of this when instantiated in a c++ code, there is no ay__en or PAD__en only ay and PAD - so no means of controlling the output enable or values.

The verilator options are:

```/usr/local/bin/verilator_bin --compiler clang --prefix Vpad_wire --Mdir /build/pads/CMakeFiles/rs_io_pads_hdl.dir/Vpad_wire.dir --make cmake --coverage --cc --trace -Wall -Wpedantic --assert -sv --hierarchical --trace --trace-structs --coverage --debug --top pad_wire hdl/pad_bidir.sv hdl/pad_wire.sv```

```/usr/local/bin/verilator_bin --compiler clang --prefix Vpad_bidir --Mdir /build/pads/CMakeFiles/rs_io_pads_hdl.dir/Vpad_wire.dir --make cmake --coverage --cc --trace -Wall -Wpedantic --assert -sv --hierarchical --trace --trace-structs --coverage --debug --top pad_wire hdl/pad_bidir.sv hdl/pad_wire.sv```

phelter, 2023-06-09:
So in the pseudocode test above:

```cpp
   m_dut->PAD = 1;
   tick();
   EXPECT_THAT(m_dut->ay, Eq(1));
```

I've traced the code and when it does the
`eval() -> eval_step() -> Vpad_wire___024root___eval()`

When it calls:
```
VL_INLINE_OPT void Vpad_wire___024root___ico_sequent__TOP__0(Vpad_wire___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vpad_wire__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vpad_wire___024root___ico_sequent__TOP__0\n"); );
    // Body
    if (((IData)(vlSelf->ay) ^ (IData)(vlSelf->pad_wire__DOT____Vtogcov__ay))) {
        vlSymsp->__Vcoverage[0].fetch_add(1, std::memory_order_relaxed);
        vlSelf->pad_wire__DOT____Vtogcov__ay = vlSelf->ay;
    }
    vlSelf->PAD = vlSelf->ay;
}
```
It changes back to `0` - so I believe when there is a situation where connecting both  `inout` together in this manner - the tristate logic to allow both directions is not added and this causes the inability to test it as a true tristate.

I was assuming it would be something like:

```c++
VL_INLINE_OPT void Vpad_wire___024root___ico_sequent__TOP__0(Vpad_wire___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vpad_wire__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vpad_wire___024root___ico_sequent__TOP__0\n"); );
    // Body
    if (((IData)(vlSelf->ay) ^ (IData)(vlSelf->pad_wire__DOT____Vtogcov__ay))) {
        vlSymsp->__Vcoverage[0].fetch_add(1, std::memory_order_relaxed);
        vlSelf->pad_wire__DOT____Vtogcov__ay = vlSelf->ay;
    }

    // Contention and tristate driver update
    std::vector<pair<uint8_t &, uint8_t &> tristatePorts = {{vlSelf->PAD, vlSelf->PAD__en}, {vlSelf->ay, vlSelf->ay__en}};
    verilatorResolveTristate(tristatePorts);
}
```

And have a  library function:

```cpp
template<typename T>
void verilatorResolveTristate (const std::vector<std::pair<T & /*port*/, uint8_t & /*en*/>> & tristatePorts)
{
    auto alreadyDriven = false;
    for(auto [&port, &en] : tristatePorts)
    {
        if (en)
        {
            if (alreadyDriven)
            {
                // Some contention detection notification or error here.
                break;
            }
            for (auto [&setPort, &setEn] : tristatePorts)
            {
                setPort = port;
             }
             alreadyDriven = true;
        }
    }
}
```

This way you could run through each of the locations and you'd need to connect up those locations.

Verilator would need to identify the I/O connection at the `TOP` - as a potential driver and then any other location within the code that drives to one of the connected points - to also have an `en` - and this particular check could be performed at the TOP to set a tristate bus within the entirety of the design based on all of the driver locations.

The only thing you'd need to create as a member of the toplevel is the list of drivers and enables of all lower block drivers.  If there are no drivers for a particular port of a submodule, i.e. a port identified as an inout port but only used as an input port - then the `&en` bit could be pointing to a constant which is always 0. This way the value of the net gets updated, but doesn't allow it to participate in the setting of the driver.

I believe this is a viable option for solving several of the tristate related Issues:
#1373, #2635 - for starters.

I'd also suggest adding a few checks:

When linting - I'd also suggest having a few tristate related linting checks:

* `inout reg` - is allowed and either be replaced with: `inout wire` or `inout tri` or `input wire` or `output logic` - this should be an error as it is not proper definition of a tristate/inout port. - And all of it's variants i.e. of where the direction vs the type are defined.
* `inout logic`  - should not be allowed and either be replaced with: `inout wire` or `inout tri`  or `input wire` or `output logic` this should be an error as it is not proper definition of a tristate/inout port. - And all of it's variants i.e. of where the direction vs the type are defined.

wsnyder, 2023-06-13:
There is currently no handling for C++ top-level inout signals, nor are there any tests that look like what you propose.  That probably contributes to why you are seeing problems ;)  So in short, debug needed.

I agree we could do resolution as you describe, and that is probably the "right" way for when there are e.g. user defined tristate resolution functions.  I would be reluctant to do that globally however as it will result in a lot more variables and so be slower than what there is now.  It's possible/likely that if we allow constant propagation and such to eliminate some of the tristate terms this would perform nearly as well (and so a win as more flexible).

The other current uglyism is the tristate resolution is done before flattening, so it does computations within each module. It would be a lot easier if done globally.

phelter, 2023-06-13:
Thanks @wsnyder.

User defined tristate resolution functions?  - No I was hoping that that could be a function call with generated code to create the list of drivers for each tristate point.

There is an addition of 1 extra driver vector per tristate signal/bus - no extra variables as each driver already comes with a `value/enable` pair already - just more pointers to pre-existing variables.  In terms of function over speed - if one is looking for speed I'd either shy away from tristate designs in verilog or systemverilog (or live with the extra cost of speed degradation if they were necessary).  There is no degradation or reduction in speed if they are not there.

Agree with the global resolution, also it would mean that the per-module would be responsible for updating/changing the `value/enable` pair but not the final resolution on the value.  At the end of the day it would remove intermediate resolution of boundaries of hierarchy which should always take on the value of the wire anyways (no enable) just a value.

wsnyder, 2026-10-21:
Closing due to age; fourstate handling when ready will resolve this.

closed 2026-10-21 by wsnyder