verilator/verilator #4283: Internal Error: V3Delayed.cpp:562: <= on read+write method

vicencb, 2023-06-09:
Hi,
when this code:
```verilog
module m();
reg c;
reg q [$];
reg x;
always @(posedge c) x <= q.pop_front();
endmodule
```
is verilated with `verilator --binary --trace m.sv` it triggers an internal error.
This happens with verilator version 5.010.
Removing the `--trace` option or changing the non-blocking assignment to a blocking one seems to fix the issue.

The documentation says: `Internal Error: This error should never occur ... , submit a bug report`, so, here it is.

wsnyder, 2023-06-13:
This works without --trace, but it seems that is just because otherwise the pop gets (mis-)optimized away.

A workaround is to use a temporary

```
always @(posedge c) begin
   y = q.pop_front();
   x <= y;
end
```

A pull to fix this case would be appreciated.  It's possible the assertion can just be removed, not sure what else will break.

wsnyder, 2024-01-26:
This was fixed by another issue in 5.016.

closed 2024-01-26 by wsnyder