verilator/verilator #4285: parameters of type struct only work if they are packed

jeras, 2023-06-10:
If I change the structure type from packed to unpacked in test `t_struct_param`:
```sh
$ git diff test_regress/t/t_struct_param.v
diff --git a/test_regress/t/t_struct_param.v b/test_regress/t/t_struct_param.v
index 8190db7b7..455989446 100644
--- a/test_regress/t/t_struct_param.v
+++ b/test_regress/t/t_struct_param.v
@@ -7,7 +7,7 @@
 `define checkd(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got=%0d exp=%0d\n", `__FILE__,`__LINE__, (gotv), (expv)); $stop; end while(0);
 
 package config_pkg;
-   typedef struct packed {
+   typedef struct {
       int         UPPER0;
       struct      packed {
          int      USE_QUAD0;
```

The test fails to compile:
```sh
$ ./test_regress/t/t_struct_param.pl 
==SUMMARY: Left 2  Passed 0  Failed 0  Time 0:00
======================================================================
dist/t_struct_param: ==================================================
-Skip: dist/t_struct_param: scenario 'dist' not enabled for test
dist/t_struct_param: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Left 1  Passed 0  Failed 0  Time 0:00
======================================================================
vlt/t_struct_param: ==================================================
	perl /home/izi/VLSI/verilator-upstream/test_regress/../bin/verilator --prefix Vt_struct_param ../obj_vlt/t_struct_param/Vt_struct_param__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_struct_param --fdedup --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_struct_param +define+TEST_DUMPFILE=obj_vlt/t_struct_param/simx.vcd t/t_struct_param.v    > obj_vlt/t_struct_param/vlt_compile.log
%Error: t/t_struct_param.v:37:42: Expecting expression to be constant, but can't determine constant for FUNCREF 'static_config'
                                : ... In instance t
        t/t_struct_param.v:23:28: ... Location of non-constant ASSIGN: LHS isn't simple variable
        t/t_struct_param.v:37:42: ... Called from static_config() with parameters:
           selector = ?32?h1
   37 |     localparam config_struct MY_CONFIG = static_config(1);
      |                                          ^~~~~~~~~~~~~
%Error: Exiting due to 1 error(s)
%Warning: vlt/t_struct_param: Exec of perl failed: %Error: t/t_struct_param.v:37:42: Expecting expression to be constant, but can't determine constant for FUNCREF 'static_config'

vlt/t_struct_param: %Error: Exec of perl failed: %Error: t/t_struct_param.v:37:42: Expecting expression to be constant, but can't determine constant for FUNCREF 'static_config'
vlt/t_struct_param: FAILED: Exec of perl failed: %Error: t/t_struct_param.v:37:42: Expecting expression to be constant, but can't determine constant for FUNCREF 'static_config'
==SUMMARY: Passed 0  Failed 1  Time 0:00

======================================================================
	#vlt/t_struct_param: %Error: Exec of perl failed: %Error: t/t_struct_param.v:37:42: Expecting expression to be constant, but can't determine constant for FUNCREF 'static_config'
		make && test_regress/t/t_struct_param.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Time 0:00
```

wsnyder, 2023-06-10:
As you noted unpacked structures are currently not supported.  A pull request would be appreciated to support this, likely needs tests and code in V3Simulate.h to process the unpacked structs and AstMemberSel operations.

samimia-swks, 2024-02-09:
I really wish I had the skills to do this, but I don't. I would highly be appreciated if this is supported.

jeras, 2025-03-18:
Hi @wsnyder I saw you have done some changes to parameter related code, including a test with assignment patterns to unpacked integer array parameters. Was there any progress on this issue? I mean I tried the latest build and it does not support packed structures for parameters, I was just wandering whether you got any new ideas about how to solve this.

wsnyder, 2025-03-18:
I haven't worked on this issue, the changes are unrelated and I can resolve any conflicts if you fix this.  This issue will mostly and perhaps entirely require changes in V3Simulate.

solomatnikov, 2025-08-22:
Any chance this could be fixed soon?

closed 2026-10-16 by wsnyder