verilator/verilator #4286: virtual interface type definition does not support parameterization

jeras, 2023-06-10:
I modified test `t_interface_virtual` to have a parameterized interface, and added the parameterization to the virtual interface type definition. The default for the parameter is not the same as the value used in the current code, this is meant to further check whether the parameter was properly propagated. Something else worth checking would be whether parameters match during assignment operations on virtual interfaces (this is buggy in Vivado 2023.1 simulator I am mostly using for reference), as described in an example toward the end of section "25.9 Virtual interfaces" of IEEE Std 1800-2017.
```diff
$ git diff test_regress/t/t_interface_virtual.v
diff --git a/test_regress/t/t_interface_virtual.v b/test_regress/t/t_interface_virtual.v
index 878df0995..0c52222c1 100644
--- a/test_regress/t/t_interface_virtual.v
+++ b/test_regress/t/t_interface_virtual.v
@@ -6,13 +6,13 @@
 
 // See also t_interface_virtual_bad.v
 
-interface PBus;
+interface PBus #(int unsigned W = 1);
    logic req, grant;
-   logic [7:0] addr, data;
+   logic [W-1:0] addr, data;
    modport phy(input addr, ref data);
 endinterface
 
-typedef virtual PBus vpbus_t;
+typedef virtual PBus #(8) vpbus_t;
 typedef vpbus_t vpbus2_t;
 
 class Cls;
```

This is the resulting error:
```sh
$ ./test_regress/t/t_interface_virtual.pl 
==SUMMARY: Left 2  Passed 0  Failed 0  Time 0:00
======================================================================
dist/t_interface_virtual: ==================================================
-Skip: dist/t_interface_virtual: scenario 'dist' not enabled for test
dist/t_interface_virtual: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Left 1  Passed 0  Failed 0  Time 0:00
======================================================================
vlt/t_interface_virtual: ==================================================
	perl /home/izi/VLSI/verilator-upstream/test_regress/../bin/verilator --prefix Vt_interface_virtual ../obj_vlt/t_interface_virtual/Vt_interface_virtual__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_interface_virtual --fdedup --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_interface_virtual +define+TEST_DUMPFILE=obj_vlt/t_interface_virtual/simx.vcd t/t_interface_virtual.v    > obj_vlt/t_interface_virtual/vlt_compile.log
%Error: Internal Error: t/t_interface_virtual.v:15:24: ../V3LinkDot.cpp:2322: Pin not under instance?
   15 | typedef virtual PBus #(8) vpbus_t;
      |                        ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
%Warning: vlt/t_interface_virtual: Exec of perl failed: %Error: Internal Error: t/t_interface_virtual.v:15:24: ../V3LinkDot.cpp:2322: Pin not under instance?

vlt/t_interface_virtual: %Error: Exec of perl failed: %Error: Internal Error: t/t_interface_virtual.v:15:24: ../V3LinkDot.cpp:2322: Pin not under instance?
vlt/t_interface_virtual: FAILED: Exec of perl failed: %Error: Internal Error: t/t_interface_virtual.v:15:24: ../V3LinkDot.cpp:2322: Pin not under instance?
==SUMMARY: Passed 0  Failed 1  Time 0:00

======================================================================
	#vlt/t_interface_virtual: %Error: Exec of perl failed: %Error: Internal Error: t/t_interface_virtual.v:15:24: ../V3LinkDot.cpp:2322: Pin not under instance?
		make && test_regress/t/t_interface_virtual.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Time 0:00
```

wsnyder, 2023-06-10:
A pull to elaborate and handle this would be appreciated.  Avoiding the specific error mentioned should be straight forward, but I don't know how much effort required beyond that to get these working.

RRozak, 2023-12-06:
I added support for parameterized virtual interfaces: https://github.com/verilator/verilator/pull/4743. Unfortunately, this example still isn't handled properly. The error message is:
```
../V3Broken.cpp:168: Broken link in node (or something without maybePointedTo): 'm_ifacep && !m_ifacep->brokeExists()' @ ./V3Ast__gen_impl.h:1183
                                                           : ... note: In instance 't'
   15 | typedef virtual PBus#(8) vpbus_t;
      |                 ^~~~
```
As I checked, the interface is marked as `dead` and removed, despite being referenced.

Jomit626, 2023-12-07:
#4743 also didn't cover virtual intferface with 2 params.
```diff
diff --git a/test_regress/t/t_interface_virtual_param.v b/test_regress/t/t_interface_virtual_param.v
index a5682a463..976cb5b3a 100644
--- a/test_regress/t/t_interface_virtual_param.v
+++ b/test_regress/t/t_interface_virtual_param.v
@@ -6,7 +6,7 @@
 // without warranty, 2015 by Todd Strader
 // SPDX-License-Identifier: CC0-1.0

-interface test_if #(parameter integer FOO = 1);
+interface test_if #(parameter integer FOO = 1, parameter integer FOO2 = 2);

    // Interface variable
    logic        data;
@@ -36,7 +36,7 @@ module t (/*AUTOARG*/
           );
    input clk;

-    virtual test_if#(.FOO(identity(5))) the_vif;
+    virtual test_if#(.FOO(identity(5)), .FOO2(4)) the_vif;

     test_if #( .FOO (identity(5)) ) the_interface ();
     test_if #( .FOO (identity(7)) ) array_interface [1:0] ();
@@ -137,6 +137,8 @@ module testmod
          $display("%%Error: intf_array[0].BAR = %0d", intf_array[0].BAR);
          $stop;
       end
+      if (intf.FOO2 != 4)
+        $stop;
       $write("*-* All Finished *-*\n");
       $finish;
    end
```
```
%Error: Internal Error: t/t_interface_virtual_param.v:39:42: ../V3Param.cpp:761: Parameter not found in sub-module: Param 'FOO2' of ''
                                                           : ... note: In instance 't'
   39 |     virtual test_if#(.FOO(identity(5)), .FOO2(4)) the_vif;
      |                                          ^~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
%Warning: vlt/t_interface_virtual_param: Exec of perl failed: %Error: Internal Error: t/t_interface_virtual_param.v:39:42: ../V3Param.cpp:761: Parameter not found in sub-module: Param 'FOO2' of ''
```

RRozak, 2023-12-07:
Yes, you're right. I will prepare a fix for a case with multiple parameters.

jeras, 2023-12-07:
I am using parameters organized into a `struct`ure, will check it when I get home from work.
I tried the build from yesterday and got some progress, but it still complains about the structured parameters.
Could you also check if `type` parameters work?

RRozak, 2023-12-08:
I unassigned myself, because I can't work on the problem with `dead` anytime soon.

wsnyder, 2026-01-19:
Also hit this with UVM tests/10resources/05seq.  Got the same error as above

`%Error: Internal Error: t/t_0.v:18:24: ../V3Broken.cpp:182: Broken link in node (or something without maybePointedTo): 'm_ifacep && !m_ifacep->brokeExists(`

This seems because the virtualized interface while used in a typedef is no where in the hierarchy, and V3LinkDot never sets user4() to say it is, then tries to delete it despite the virtual typedef reference.

Test case:
```
interface mem_if #(
    int unsigned ADDR_SIZE = 16
);
endinterface

module t;
  class Cls #(
      type T = int
  );
  endclass

  typedef Cls#(virtual mem_if #(8)) mem_if_t;
endmodule
```