verilator/verilator #4287: trouble understanding "Unsupported: virtual interface never assigned any actual interface"

jeras, 2023-06-10:
I am getting this error, which I am unable to fix. I do not think there is anything wrong with the code, I create an a virtual interface instance and assign it an actual interface in the constructor. The [code](https://github.com/jeras/TCB/blob/main/hdl/tbn/vip/tcb_vip_pkg.sv#L58-L83) (system bus driver) is working in Verilator. I followed this [guide](https://verificationguide.com/systemverilog/systemverilog-virtual-interface/).
```sh
%Error-UNSUPPORTED: ../../hdl/tbn/vip/tcb_vip_pkg.sv:60:13: Unsupported: virtual interface never assigned any actual interface
                                                          : ... In instance tcb_lib_endianness_tb.dut
   60 |     virtual tcb_if tcb;
      |             ^~~~~~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.011
%Error-UNSUPPORTED: ../../hdl/tbn/vip/tcb_vip_pkg.sv:64:47: Unsupported: virtual interface never assigned any actual interface
                                                          : ... In instance tcb_lib_endianness_tb.dut
   64 |     function new(string MODE = "MON", virtual tcb_if tcb);
      |                                               ^~~~~~
%Error: Exiting due to 2 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Could you provide me with some insight on the meaning of this error, so I can fix my code or create a proper bug report.
I compared my virtual interface code with the passing regression test `t_interface_virtual`, but I can't figure what I am doing wrong.

I encountered this issue a few months ago, I am not sure what I did then, but I think I just waited for Vivado to release a new version so I could make some progress. Now that I encountered the next blocking issue in Vivado I tried again in Verilator.

I will try to strip down my code to a regression test.

wsnyder, 2023-06-10:
Verilator thinks there is a virtual interface that either is never assigned to a non-virtual interface, or was assigned to a non-virtual interface that later got deleted (perhaps due to parameters in the interface).  I would first suspect something related to parameter resolution.

wsnyder, 2023-08-12:
Do you have an example for this?

jeras, 2023-08-12:
Hi, I do not have an example, and the code causing it was recoded a few time since then, I will close this for now.

closed 2023-08-12 by jeras