verilator/verilator #4289: How can I make verilator/compile even faster?

PeterMonsson, 2023-06-12:
How may we help - what is your question?

I am trying to get the incremental verilation/compile step as fast as possible. 

I use ccache as well as the following command line options -j 0 -CFLAGS -O0 to speed things up (g++ 12.2.0).

This works pretty well. A single line change for a single file is 1.3seconds, but a single line change in multiple files grows to 3.7seconds (all files are systemverilog)

I am wondering if there is anything that I can do to make incremental compile faster?

For example, maybe I could add output-split to the end of a package?

wsnyder, 2023-06-13:
Output split is turned on by default, you could use a lower setting.

When you say "in multiple files" what is an example change, and what output files are changing.

poemonsense, 2023-06-13:
using clang++ instead of g++ greatly benifits the compile time, from <1 hours to several minutes

wsnyder, 2023-06-13:
Some versions of g++ are faster than clang, depends on which versions are involved.  If there is a very big difference there is likely a compiler bug.

Also use mold for linking.  Verilator master now does this if mold is installed.

udif, 2023-06-13:
I was investigating recently the same issue, and came to a similar conclusion (clang is compiler much faster).
The point is that compiler comparison is always based on what's available on a given platform:

For CentOS 7 (my eventual target) there are no relevant C++20 compilers, so I don't build on it.
For Ubuntu 20.04 (my current build platform) it's clang-12 vs gcc-10, which I found clang to be have much faster compile times (and better runtime as well IIRC).
For Ubuntu 22.04 (not available in my current setup), I think it's clang-15 vs gcc-12, which I haven't benchmarked.

PeterMonsson, 2023-06-13:
Hi all, 

Thank you for your suggestions. I have tried playing with output-split, but this did not give me any improvements. I think that it is quite hard to get right. --output-split 1000 does nothing and --output-split 700 many independent files that get recompiled even on an incremental compile.

I have also tried to use clang, but I did only see a 0.1second difference for my setup.

If you want to try it out yourself, I am running SVUnit on PeterMonsson/sv_waveterm with the following command:

    (cd tests; time runSVUnit -c_arg "-I../src" -c_arg "-j 0" -c_arg "-CFLAGS -O0")

I trigger an incremental compile by adding 

    $display("new string");

to line 193 of tests/sv_waveterm_unit_test.sv

gezalore, 2023-06-13:
I'm not sure how serious this is but based on you being willing to use `-O0` I guess you don't really care about the model speed. In that case, you can turn off module inlining. The sledgehammer for this is `verilator -fno-inline`. This is likely to hurt actually as it will likely generate a lot more output files, but in single threaded mode (--threads 1) this will minimize incremental C++ builds (or it did last time I tried 3 years ago). A more production worthy attempt is to selectively turn off inining for decent sized chunks of your design hierarcy using `// verilator no_inline` or the equivalent directive in a vlt config file.

With -O0, I'm not surprised you see no difference between gcc/clang and their versions, but just in case use `-g0` on your C++ compiler. With -O0 you are quite possibly IO bound so work from ramdisk (your ccache and input sv, output C++ files) if you are willing to go to silly extents.

PeterMonsson, 2023-06-14:
Thank you for your suggestions @gezalore, but they did not do anything meaningful for me. Both -fno-inline and -g0 with clang++ (version 14.0.6) give me about 3.4seconds. 

The largest step in the flow for incremental compile is this step:

    ccache clang++ -Os  -I.  -MMD -I/home/petermonsson/chips/verilator/include -I/home/petermonsson/chips/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O0 -g0 -DVL_TIME_CONTEXT   -fcoroutines-ts -c -o Vtestrunner__ALL.o Vtestrunner__ALL.cpp

Is there a way for me to control verilator_includer so that it emits chunks of Vtestrunner__ALL.cpp?

gezalore, 2023-06-14:
try using VM_PARALLEL_BUILDS=1 at build time (this is on by default if output split actully split a file), but you can add it yourself to your `make` line. See `include/verilated.mk`

wsnyder, 2023-06-17:
Closing as seems answered but feel free to followup.

closed 2023-06-17 by wsnyder