verilator/verilator #4290: Why is verilator using m_threads instead of modelp->threads()?

poemonsense, 2023-06-13:
If I'm not missing something, PR #3477 introduces a deviation from the previous version.

Instead of using the model's threads, verilator is now using the same threads as the physical machine: `if (!m_threadPool) m_threadPool.reset(new VlThreadPool{this, m_threads - 1});` where `unsigned m_threads = std::thread::hardware_concurrency();` On my machine with 24 physical threads, `VlThreadPool` has 23 threads in `m_workers` even if the model has only one thread.

It does bring functional issues but I think it's just unnecessary to have this.

wsnyder, 2023-06-13:
I believe the reason for this is you can have multiple Verilated models present in a simulation, (e.g. you may have 3 models in a process each using 2 threads each), and the thread pool needs to support that by having a thread ready for each physical core. I.e. the thread pool on a 20 core machine will have 19 threads (plus master).

closed 2023-06-13 by wsnyder

poemonsense, 2023-06-14:
I thought every model should have their own thread pool and control their own threads.