verilator/verilator #4294: Fix docker build

wsnyder, 2023-06-14:
I just made a release, first one since @jlloyola docker fixes in #4163 (thanks).

@jlloyola, the build returned an error would you mind taking a look please?

https://github.com/verilator/verilator/actions/runs/5261396685

```
Buildx version
  /usr/bin/docker buildx version
  github.com/docker/buildx 0.10.4+azure-1 c513d34049e499c53468deac6c4267ee72948f02
/usr/bin/docker buildx build --build-arg SOURCE_COMMIT=v5.012 --iidfile /tmp/docker-actions-toolkit-8lvKVP/iidfile --label org.opencontainers.image.created=2023-06-13T23:37:44.188Z --label org.opencontainers.image.description=Verilator open-source SystemVerilog simulator and lint system --label org.opencontainers.image.licenses=LGPL-3.0 --label org.opencontainers.image.revision=7d2d32420a630befa4097170ecbf227e04e32522 --label org.opencontainers.image.source=https://github.com/***/*** --label org.opencontainers.image.title=*** --label org.opencontainers.image.url=https://github.com/***/*** --label org.opencontainers.image.version=v5.012 --platform linux/arm64,linux/amd64 --provenance mode=max,builder-id=https://github.com/***/***/actions/runs/5261396685 --tag ***:v5.012 --tag ***:latest --tag /***:v5.012 --tag /***:latest --metadata-file /tmp/docker-actions-toolkit-8lvKVP/metadata-file --push ci/docker/run
ERROR: invalid tag "/***:v5.012": invalid reference format
Error: buildx failed with: ERROR: invalid tag "/***:v5.012": invalid reference format
```

jlloyola, 2023-06-14:
Ack. Will take a look

jlloyola, 2023-06-21:
Had some cycles to look at this today. I suspect the issue is related to secret masking:
https://github.com/verilator/verilator/actions/runs/5261396685/jobs/9509451871#step:5:3
I'll refactor the action to define `DOCKER_HUB_NAMESPACE` as a repo variable instead of a secret (no need for it to be a secret really). The only reason I made it a secret was because `act` did not support repo variables (This has been addressed in https://github.com/nektos/act/issues/1558)

wsnyder, 2023-12-10:
Fixed earlier with #4419

closed 2023-12-10 by wsnyder